trezor/trezor-firmware #2454: Model R: EIP712 signing flow

grdddj, 2022-08-17:
Currently, some of the features are missing - for example, the possibility to cancel viewing a field while scrolling, or the overall possibility to cancel it at the beginning (as uncovered by device tests).

It goes mostly together with the `should_show_more` functionality.

There is also a question of whether to support EIP712 full-data viewing or not - as on `model 1` we have only blind signing because of screen-size limitations, and `model R` has the same issue.

aloisklink, 2022-08-26:
> as on `model 1` we have only blind signing because of screen-size limitations

Off the top of my head, I believe that the reason why `model 1` only supports blind signing was was because it was very difficult (maybe even impossible) to decode the EIP-712 data structures in C.

I vaguely remember somebody saying that the C protobuf library didn't support recursive protobuf types, but it was a while ago, so I'm not 100% sure if that's accurate.

> There is also a question of whether to support EIP712 full-data viewing or not

I'd say it's still really important for anybody in the Ethereum space, because there are a lot of tokens that can be taken from your account automatically if you sign the wrong EIP-712 transaction.

If somebody hacks your PC, they could do a MITM attack and when you think you're doing a simple EIP-712 signing (e.g. for voting), they could instead send an EIP-712 for "Allow attackers address to spend INFINITY of token X".

Hannsek, 2023-01-13:
[image]

grdddj, 2023-01-24:
Third "INFO" button added in [5b5d5b7](https://github.com/trezor/trezor-firmware/pull/2610/commits/5b5d5b7043bf69edd0c6e4c6f810c181c8487a09)

grdddj, 2023-01-24:
Changed `ALL` button to `i`, and it really looks better - [0c1e598](https://github.com/trezor/trezor-firmware/pull/2610/commits/0c1e598f90a73bbf235a738477ad7e25a6d53794)

[image]

closed 2023-05-31 by grdddj