trezor/trezor-firmware #2486: Extend decimals of fee rate to 2 digits

hynek-jina, 2022-09-06:
- Trezor Suite allows you to set two decimal places for the fee rate. https://github.com/trezor/trezor-suite/issues/3252
- Trezor is currently rounding received input to only one decimal which might be confusing for the user.

**Proposed solution:**
- Extend decimals of fee rate to 2 digits
- Maybe show them only if they are non-zero? (like `2.00` → `2`, but `2.01` keep `2.01`)

hynek-jina, 2022-09-06:
> i think that's what is happening now?

Nice, then it is only about extension to two decimals.. Because currently it shows only one..

closed 2022-09-09 by matejcik

STew790, 2022-09-20:
**QA OK**
Both T1 and TT are correctly showing 2 digit decimals in fee rate.

**Info:**
- Suite version: web 22.9.2 (cfeba9a3533ec7f4a5d4f31da4ac8c8b768547ed)
- Browser: Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/105.0.0.0 Safari/537.36
- OS: Linux x86_64
- Screen: 1920x1080
- Device: model T 2.5.3 Universal (revision https://github.com/trezor/trezor-firmware/commit/4bed278e80d23077676128eba8cb2478fcd31120)
- Device: model 1 1.11.3 Universal (revision https://github.com/trezor/trezor-firmware/commit/4bed278e80d23077676128eba8cb2478fcd31120)
- Transport: bridge 2.0.32