trezor/trezor-firmware #2493: tests: --ui=record does not work with multicore testing

matejcik, 2022-09-07:
AFAICT each runner tries to write its own results, they all write over each other, and the changes are lost. My understanding is that this is essentially random and _some subset_ of changed tests _would_ get written out given the right timing.

In any case, each runner must write their own fixture results, and the xdist controller is the one to merge them into the resulting fixtures.json.

General improvement suggestion: it would be nice if, internally, each test run produced a separate list of ui hashes (then the workers can dump that into `fixtures.gwXYZ.json`), and this list would then be _merged_ into the main list, keeping or removing pre-existing tests based on `--ui-check-missing`

grdddj, 2023-01-20:
Do we still want this, or is the current solution of 
```
test_emu_ui_record_multicore: ## quickly record all screens
	make test_emu_ui_multicore || echo "All errors are recorded in fixtures.json"
	make test_emu_accept_fixtures
```

enough? In that case we could close this

matejcik, 2023-01-24:
I think so, yes.

I was also thinking we could completely remove the `record` option at least internally, in favor of always running as if `test` and possibly running the "accept" code path afterwards.

closed 2023-01-24 by matejcik