trezor/trezor-firmware #2500: altcoins

hynek-jina, 2022-09-08:


grdddj, 2022-09-13:
Below are the custom screens in their current look.

To make sure, the goal here is to make it look exactly the same, or just "somehow make it work"?

[image]
[image]
[image]
[image]
[image]

grdddj, 2022-10-05:
Done in https://github.com/trezor/trezor-firmware/pull/2545

closed 2022-10-24 by mmilata