trezor/trezor-firmware #2501: trezor emulator build failure on linux: ‘ed25519_hram’ reading 64 bytes from a region of size 32

donpdonp, 2022-09-08:
**Describe the bug**
A clear and concise description of what the bug is.

**Firmware version and revision**
git head

**Desktop/smartphone setup (please complete the following information):**
ubuntu 22.04 on intel thinkpad 

**To Reproduce**
Steps to reproduce the behavior:
$ make build_unix
vendor/trezor-crypto/ed25519-donna/ed25519.c: In function ‘ed25519_cosi_sign’:
vendor/trezor-crypto/ed25519-donna/ed25519.c:86:9: error: ‘ed25519_hram’ reading 64 bytes from a region of size 32 [-Werror=stringop-overread]
   86 |         ed25519_hram(hram, R, pk, m, mlen);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
vendor/trezor-crypto/ed25519-donna/ed25519.c:86:9: note: referencing argument 2 of type ‘const unsigned char *’
vendor/trezor-crypto/ed25519-donna/ed25519.c:34:1: note: in a call to function ‘ed25519_hram’
   34 | ed25519_hram(hash_512bits hram, const ed25519_signature RS, const ed25519_public_key pk, const unsigned char *m, size_t mlen) {
      | ^~~~~~~~~~~~
cc1: all warnings being treated as errors
scons: *** [build/unix/vendor/trezor-crypto/ed25519-donna/ed25519.o] Error 1
make: *** [Makefile:127: build_unix] Error 2

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

prusnak, 2022-09-08:
Which commit are you on? HEAD is not enough

donpdonp, 2022-09-08:
```
$ git log -1
commit 18c3a6a5bd45923380312b064be96155f5a7377d (HEAD -> master, origin/master, origin/HEAD)
Author: Tomas Susanka <tsusanka@gmail.com>
Date:   Fri Apr 19 16:21:15 2019 +0200

    README: we're moving to monorepo
donp@carbon trezor-core $ git status
On branch master
Your branch is up to date with 'origin/master'.
```

donpdonp, 2022-09-08:
removing -Werror is a temporary workaround

```
diff --git a/SConscript.unix b/SConscript.unix
index b5f53d0c..dafcf52f 100644
--- a/SConscript.unix
+++ b/SConscript.unix
@@ -316,7 +316,7 @@ env.Replace(
     COPT=env.get('OPTIMIZE', env.get('ENV').get('OPTIMIZE', '-Os')),
     CCFLAGS='$COPT '
     '-g3 '
-    '-std=gnu99 -Wall -Werror -Wuninitialized '
+    '-std=gnu99 -Wall -Wuninitialized '
     '-fdata-sections -ffunction-sections ' + CCFLAGS_MOD,
     CCFLAGS_QSTR='-DNO_QSTR -DN_X64 -DN_X86 -DN_THUMB',
     LIBS=['m'] + LIBS_MOD,
```

prusnak, 2022-09-08:
> $ git log -1

You are on a 3-year old commit of a no longer active repository (trezor-core). Please close this repository instead (`git clone --recursive https://github.com/trezor/trezor-firmware.git).

closed 2022-09-08 by prusnak

donpdonp, 2022-09-08:
Okay thanks. This repo is still the second hit in google for 'trezor emulator'