trezor/trezor-firmware #2512: Rust UI: introduce Icon type

matejcik, 2022-09-14:
The code especially in #2471 is getting a little messy in places because we don't have a convenient `ui::Icon` type with `size()` and `pub data: &'static [u8]` and similar methods, which need to be laboriously explicitly written out in all usage sites.

Let's not block the above PR for that, but we should introduce `Icon` and factor out these things everywhere.

I believe there's some work on that in #2277 already?

grdddj, 2022-09-14:
> I believe there's some work on that in https://github.com/trezor/trezor-firmware/pull/2277 already?

Yeah, I have defined an `Icon` struct - https://github.com/trezor/trezor-firmware/blob/7a938de855d03e173e40bed8c30b0835b42a9ca4/core/embed/rust/src/ui/display.rs#L1066-L1131

The global objects are then `IconAndName` - https://github.com/trezor/trezor-firmware/blob/7a938de855d03e173e40bed8c30b0835b42a9ca4/core/embed/rust/src/ui/model_tr/theme.rs#L31-L65

matejcik, 2022-09-14:
to resolve this issue, `Icon` type should include the same fields as `ToifInfo`, in addition to the `data` cut off at the right point (i.e. a slice of `toif_bytes[12..]`)

(also probably should not be `Copy` unless there is a strong reason)

PR #2708 (2023-01-30): Introduce Toif / Icon / Image type
Introduces Icon and Image types, and more general Toif type.

Did this without `Copy`, but its a bit annoying at some places, so i left it as a fixup, see the difference with and without here: https://github.com/trezor/trezor-firmware/commit/cb2f10035829d7c82612a5132f336f89349dda43 

closes #2512
closed 2023-01-30 by matejcik