trezor/trezor-firmware #2513: T1 bootloader: verify firmware signatures based on SignMessage

matejcik, 2022-09-14:
The T1 bootloader verifies the firmware signature via a manual ECDSA 3-of-5 scheme:

1. let the message `m` be the firmware header with signature slots zeroed out.
   * (implementation-wise, take a SHA256 digest of the above data)
2. for each of the 3 signature slots:
    1. verify that the pubkey index points to one of the 5 baked in public keys
    2. verify that the pubkey index is different from the other two slots, i.e., all 3 slots are signed by different public keys
    3. verify that this is a valid signature of `m` for the selected pubkey
        * (implementation-wise, we call `ecdsa_verify_digest` instead of `ecdsa_verify`, because we already calculated the digest)

This entails signing arbitrary data with ECDSA, which cannot be done on Trezor directly.

We want to change the signing method so that it is **possible to verify the digest being signed on Trezor screen**.
The signer will calculate SHA256 digest of `m` (fw header minus signature slots) on host and send it to Trezor as data for Bitcoin `SignMessage`.
Trezor will display the digest on screen, and when confirmed, produce a signature of the following data:
`\x18Bitcoin Signed Message:\n\x20<32 bytes of digest>`.

On the bootloader side, the following is the updated algorithm:
1. let `m'` be the firmware header with signature slots zeroed out
2. let `m` be `"\x18Bitcoin Signed Message:\n\x20" + SHA256(m')`
3. continue with (2) from the original algorithm, i.e., the signatures are now verifying the changed `m`

hiviah, 2022-09-23:
This leaves a question which hash should now be displayed both at startup or update.

After reviewing possible paths of update/start/wipe combinations it makes probably most sense to show the `m'` hash - which is the same as shown with current scheme (confusingly called "new" in code).

Some points:

1. It will be compatible with the current hash computation, so there will be no need for Suite to make extra special cases
2. Hash shown of the same FW will be thus same before and after update (including various paths like wipe/update cycles)

matejcik, 2022-09-26:
you are absolutely correct, also showing hash of `m'` means showing the same thing the signer is confirming on Trezor

hiviah, 2022-09-28:
One note: because signing with Bitcoin SignMessage does double SHA-256, the value signed output from SignMessage and checked by FW is:

`SHA256("\x18Bitcoin Signed Message:\n\x20" + binary_32_bytes_of digest)`

`x18` is from coin info and `0x20` is the length following it, 32 bytes.

Also output from SignMessage needs have its first byte dropped to match 64-byte secp256k1 format used by T1 FW.

matejcik, 2022-09-28:
updated the algorithm in summary

hiviah, 2022-10-03:
Short PoC to generate 5 signatures with 5 keys for some testing reference.

**Important** do not use the hash shown by cibuild, but the one shown on T1 screen on install or shown by `trezorctl firmware-update`!

```
#!/usr/bin/env python3
import sys
from hashlib import sha256
from pathlib import Path
from trezorlib.client import get_default_client
from trezorlib.btc import sign_message, get_public_node
from trezorlib.tools import parse_path
from binascii import *

client = get_default_client()
# arg1 is SHA256 digest as reported by "trezorctl firmware-update -f trezor.bin", not the one shown on cibuild!
# The same should be shown on T1 screen
digest = unhexlify(sys.argv[1])
print("Hash before sig: ", digest.hex())
assert(len(digest) == 32)

for i in range(0, 5):
    print(f"--- Key/sig {i}")
    path_text = f"m/44'/0'/{i}'/0/0"
    ADDRESS_N = parse_path(path_text)
    print("Addres_n", path_text, ADDRESS_N)

    node = get_public_node(client, ADDRESS_N)
    print("Public key:", node.node.public_key.hex())
    print("xpub:", node.xpub)

    signature = sign_message(client, "Bitcoin", ADDRESS_N, digest)
    sig_64bytes = signature.signature[1:] #first byte stripped to match normal secp256k1
    assert (len(sig_64bytes) == 64)
    print("Signature:", sig_64bytes.hex())
```

hiviah, 2022-10-03:
A sanity check script the scheme works as expected using different library/implemetation:

```
#!/usr/bin/env python3
import ecdsa
from hashlib import sha256
import sys

# arg 1 - hex digest of firmware header with zeroed sigslots
# arg 2 - public key (compressed or uncompressed)
# arg 3 - signature 64 bytes in hex
digest = bytes.fromhex(sys.argv[1])
assert len(digest) == 32
public_key = bytes.fromhex(sys.argv[2])
sig = bytes.fromhex(sys.argv[3])
assert len(sig) == 64
# 0x18 - coin info, 0x20 - length of digest following
prefix = b"\x18Bitcoin Signed Message:\n\x20"
message_predigest = prefix + digest
message = sha256(message_predigest).digest()

vk = ecdsa.VerifyingKey.from_string(public_key, curve=ecdsa.SECP256k1, hashfunc=sha256)
result = vk.verify(sig, message)
print("Signature verification result", result)
```

E.g. (pubkeys and signatures were printed with the signing script in above comment):

`./firmware_hash_verify.py 0029608e2c879b6f6f636faba08d3434319e4694f6e0cdd626e4216640c403a1 02e62488c7a4aee638457f8b6afdb6dc41993971b5175dd4b941e14441c2df22b2 1ce267b37d712776f856ace87c8633413699b7212c63577fcac654c91ae9d35347f208fc7fb6d2315514a111c3d951a2c6451cc2a6178bdf12dc0c76406a6c08` - works

Changing hashing or changing values will show that it fails when it should fail.

hiviah, 2022-10-04:
Helper script to binary patch signatures directly into `trezor.bin`

[fill_t1_fw_signatures.py.txt](https://github.com/trezor/trezor-firmware/files/9707774/fill_t1_fw_signatures.py.txt)

Note: FW patched this way must be flashed via SWD (trezorctl will refuse the signature scheme ATM).

matejcik, 2022-10-14:
1. we need an _intermediary firmware_ signed 3 ways:
   * v3 - new-style header with SignMessage keys
   * v2 - new-style header with original keys
   * v1 - old-style header as a prefix of the v3 image
2. we will distribute two binaries:
   * v1 header + v3 header + code
      * for bootloaders \< 1.8.0, the v1 header is valid and will allow installation
      * for new bootloaders that recognize v3 signatures, we strip away the v1 header and install the "inner" image
   * v2 header + code
      * for bootloaders >= 1.8.0, the v2 header is valid and will allow installation
3. the bootloader **does not need** to understand v1 headers, because there is no situation in which contents of flash do not have a valid new-style v2 or v3 header
4. for purpose of **validating the existing image** that is on flash, the bootloader must accept both v2 and v3 style signatures
5. for purpose of **installing new firmware**, the bootloader must only accept v3 style signatures
    * this means that it will not be possible to use the old keys to sign new firmwares, which is what we want.

## Variants:

### 1. installing on bootloader \< 1.8.0

Old bootloader will accept v1-signed image.
The data on flash is:
1. storage sector containing the v1 header (+ seed, if present)
2. application sectors containing the v3 header + code

Old bootloader will boot the v3 firmware via the `header_len` trick (the second field of new-style header is in fact a pointer to `reset_handler()`, exactly where it _would_ be in an old style v1 image).
The firmware installs the new bootloader.
New bootloader sees the v3 header with valid signatures exactly where it should be, and ignores the v1 header in the storage sector.

### 2. installing on bootloader >= 1.8.0

Old bootloader will accept v2-signed image.
The data on flash is:
1. storage sector containing seed, if present
2. application sectors containing the v2 header + code

Old bootloader will boot the firmware, because v2 and v3 headers are in fact identical and the VTOR is at the same place.
The firmware installs the new bootloader.
New bootloader sees v2 header. For this reason we require point (4) above: new bootloader must accept this image as valid, otherwise it would erase storage.

### 3. Installing on itself (bootloader 1.12.0 or however much)

New bootloader will accept v3-signed image.
The data on flash is:
1. storage sector containing seed, if present
2. application sectors containing the v3 header + code

New bootloader will boot the firmware natively.
The firmware will see that the bootloader is up to date.

matejcik, 2022-10-17:
Some device-tests do, but that won't be a problem if you keep the `signatures_new_ok()` as-is.
This was actually introduced very recently.

hiviah, 2022-10-17:
Implemention being prepared in PR #2568 .

matejcik, 2022-11-22:
let's keep this open because we still need to update the code with the new keys

matejcik, 2022-12-14:
As part of this change, we need to lower the quorum, from 3 of 5 to *2 of 3* public keys.

PR #2568 did not do that, @TychoVrahe can you prepare the relevant change? We should be getting the production keys Soon(tm).

prusnak, 2022-12-14:
> We should be getting the production keys Soon(tm).

Monday 19th December at ~noon (12:00) to be exact.

closed 2023-01-04 by hynek-jina