trezor/trezor-firmware #2515: Remove all the dead code

grdddj, 2022-09-14:
With the help of [Vulture tool](https://pypi.org/project/vulture/), I was able to find some dead code in `core` codebase that gets included into the binary and takes space.

It could be an easy win of deleting these - at least most of them. All the "dead" symbols below make around `3 kb`

```
.flash2    203     3    L203     D0       src/trezor/crypto/slip39.py:414                              _rs1024_error_index()
.flash2    99      6    L99      D0       src/apps/cardano/layout.py:78                                is_printable_ascii_bytestring()
.flash2    30      3    L30      D0       src/apps/cardano/sign_tx/signer.py:256                       Signer._should_show_tx_hash()
.flash2    27      3    L27      D0       src/apps/cardano/sign_tx/plutus_signer.py:48                 PlutusSigner._should_show_tx_hash()
.flash2    84      3    L84      D0       src/apps/common/writers.py:22                                write_uint16_be()
.flash2    150     6    L102     D48      src/apps/management/recovery_device/layout.py:92             show_dry_run_different_type()
.flash2    60      4    L44      D16      src/apps/monero/signing/offloading_keys.py:84                hmac_key_txout_asig()
.flash2    61      4    L45      D16      src/apps/monero/signing/offloading_keys.py:105               enc_key_cout()
.flash2    68      4    L52      D16      src/apps/monero/signing/offloading_keys.py:119               det_comm_masks()
.flash2    69      3    L69      D0       src/apps/monero/xmr/addresses.py:32                          decode_addr()
.flash2    60      3    L60      D0       src/apps/monero/xmr/addresses.py:42                          public_addr_encode()
.flash2    102     3    L102     D0       src/apps/monero/xmr/bulletproof.py:206                       _sc_mul8()
.flash2    67      3    L67      D0       src/apps/monero/xmr/bulletproof.py:458                       KeyVBase.slice()
.flash2    390     4    L374     D16      src/apps/monero/xmr/bulletproof.py:637                       KeyV.realloc_init_from()
.flash2    85      3    L85      D0       src/apps/monero/xmr/bulletproof.py:701                       KeyVSized
.flash2    59      3    L59      D0       src/apps/monero/xmr/bulletproof.py:709                       KeyVSized.__init__()
.flash2    48      3    L48      D0       src/apps/monero/xmr/bulletproof.py:716                       KeyVSized.__setitem__()
.flash2    42      3    L42      D0       src/apps/monero/xmr/bulletproof.py:713                       KeyVSized.__getitem__()
.flash2    44      3    L44      D0       src/apps/monero/xmr/bulletproof.py:1310                      _const_vector()
.flash2    103     3    L103     D0       src/apps/monero/xmr/bulletproof.py:1503                      _vector_gen()
.flash2    68      3    L68      D0       src/apps/monero/xmr/bulletproof.py:2394                      BulletProofPlusBuilder.vector_exponent()
.flash2    114     3    L114     D0       src/apps/monero/xmr/crypto_helpers.py:122                    generate_signature()
.flash2    193     4    L177     D16      src/apps/monero/xmr/crypto_helpers.py:140                    check_signature()
.flash2    60      3    L60      D0       src/apps/monero/xmr/serialize/__init__.py:33                 dump_msg_gc()
.flash2    39      2    L39      D0       src/apps/monero/xmr/serialize/base_types.py:67               UInt8
.flash2    74      3    L74      D0       src/apps/monero/xmr/serialize/int_serialize.py:45            load_uvarint_b()
.flash2    75      3    L75      D0       src/apps/monero/xmr/serialize/int_serialize.py:82            dump_uint_b_into()
.flash2    67      3    L67      D0       src/apps/monero/xmr/serialize/message_types.py:20            UnicodeType
.flash2    46      3    L46      D0       src/trezor/log.py:55                                         critical()
.flash2    88      3    L88      D0       src/trezor/strings.py:24                                     format_ordinal()
.flash2    98      2    L98      D0       src/trezor/ui/__init__.py:108                                click()
.flash2    67      3    L67      D0       src/trezor/utils.py:195                                      BufferWriter
.flash2    92      3    L92      D0       src/trezor/utils.py:211                                      BufferWriter.write()
.flash2    64      3    L64      D0       src/trezor/utils.py:202                                      BufferWriter.seek()
.flash2    42      3    L42      D0       src/trezor/utils.py:198                                      BufferWriter.__init__()
.flash2    77      4    L61      D16      src/trezor/workflow.py:100                                   kill_default()
```

Some local variables that are not existing as standalone symbols - these should also free some space:
```
core/src/apps/monero/xmr/clsag.py:57: unused variable 'KeyM'
core/src/apps/cardano/helpers/bech32.py:16: unused variable 'HRP_STAKE_SHARED_KEY_HASH'
core/src/apps/monero/xmr/serialize/base_types.py:40: unused class 'XmrStructuredType'
core/src/apps/monero/xmr/serialize_messages/base.py:16: unused variable 'DATA_ATTR'
core/src/apps/monero/xmr/serialize_messages/base.py:23: unused variable 'DATA_ATTR'
core/src/apps/monero/xmr/serialize_messages/base.py:29: unused variable 'ECPublicKey'
core/src/apps/monero/xmr/serialize_messages/tx_prefix.py:10: unused variable 'VARIANT_CODE'
core/src/apps/stellar/consts.py:79: unused variable 'ASSET_TYPE_NATIVE'
core/src/apps/stellar/consts.py:80: unused variable 'ASSET_TYPE_ALPHANUM4'
core/src/apps/stellar/consts.py:81: unused variable 'ASSET_TYPE_ALPHANUM12'
core/src/apps/tezos/helpers.py:75: unused variable 'OP_TAG_ENDORSEMENT'
core/src/apps/tezos/helpers.py:76: unused variable 'OP_TAG_SEED_NONCE_REVELATION'
core/src/apps/tezos/helpers.py:77: unused variable 'OP_TAG_DOUBLE_ENDORSEMENT_EVIDENCE'
core/src/apps/tezos/helpers.py:78: unused variable 'OP_TAG_DOUBLE_BAKING_EVIDENCE'
core/src/apps/tezos/helpers.py:79: unused variable 'OP_TAG_ACTIVATE_ACCOUNT'
core/src/apps/webauthn/fido2.py:1170: unused attribute 'versionInterface'
core/src/apps/webauthn/fido2.py:1171: unused attribute 'versionMajor'
core/src/apps/webauthn/fido2.py:1172: unused attribute 'versionMinor'
core/src/apps/webauthn/fido2.py:1173: unused attribute 'versionBuild'
core/src/apps/webauthn/fido2.py:1174: unused attribute 'capFlags'
core/src/apps/webauthn/fido2.py:1257: unused attribute 'registerId'
core/src/apps/webauthn/fido2.py:1259: unused attribute 'keyHandleLen'
core/src/trezor/log.py:6: unused variable 'NOTSET'
core/src/trezor/ui/style.py:18: unused variable 'PINK'
core/src/trezor/ui/style.py:20: unused variable 'DEEP_PURPLE'
core/src/trezor/ui/style.py:21: unused variable 'INDIGO'
core/src/trezor/ui/style.py:23: unused variable 'LIGHT_BLUE'
core/src/trezor/ui/style.py:24: unused variable 'CYAN'
core/src/trezor/ui/style.py:25: unused variable 'TEAL'
core/src/trezor/ui/style.py:27: unused variable 'LIGHT_GREEN'
core/src/trezor/ui/style.py:28: unused variable 'LIME'
core/src/trezor/ui/style.py:30: unused variable 'AMBER'
core/src/trezor/ui/style.py:32: unused variable 'DEEP_ORANGE'
core/src/trezor/ui/style.py:33: unused variable 'BROWN'
core/src/trezor/ui/style.py:37: unused variable 'BLUE_GRAY'
core/src/trezor/wire/codec_v1.py:20: unused variable 'INVALID_TYPE'
```

grdddj, 2022-09-22:
Done as a part of https://github.com/trezor/trezor-firmware/pull/2525

matejcik, 2022-11-14:
fixed in #2525

closed 2022-11-14 by matejcik