trezor/trezor-firmware #2532: FIDO layouts

hynek-jina, 2022-09-29:


closed 2022-11-24 by matejcik

bosomt, 2022-12-14:
**QA OK**

Works as expected /tested on Google and GitHub account/
Noted one thing, that register / authenticate message and web log/name is not properly aligned but its probably expected /it matches screenshots from PR/

[image]

[image]

**Info:**
 - Suite version: desktop 22.12.1 (ddaacdb2245d7a909b3d2abfe918cff34d2e1609)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/22.12.1 Chrome/104.0.5112.124 Electron/20.3.5 Safari/537.36
 - OS: MacIntel
 - Screen: 1440x900
 - Device: model T 2.5.4 Universal (revision ff35c77bad82c7acac1df6245940ebe0d6454642)
 - Transport: bridge 2.0.32

Hannsek, 2022-12-14:
@mmilata Did you fix the layout? (https://satoshilabs.slack.com/archives/C02BY51RD8R/p1669232972198689)

mmilata, 2022-12-14:
No

Hannsek, 2022-12-14:
Then please fix the layout.