trezor/trezor-firmware #2540: Fix `ticks_ms` arithmetic

andrewkozlik, 2022-10-04:
In several places we do arithmetic operations `+`, `-` and `<` with the values returned by `utime.ticks_ms()`. These values are not meant to be handled directly and doing so may produce incorrect results. Instead we should be using `utime.ticks_add()` and `utime.ticks_diff()`, see [documentation](https://github.com/trezor/micropython/blob/fce614cbe213e3f6c9e1a2a9634d2c1edae7c57b/docs/library/time.rst).

The places where we handle the values incorrectly are:
apps/common/request_pin.py
apps/homescreen/homescreen.py
apps/webauthn/fido2.py
trezor/ui/loader.py

naiyoma, 2024-02-29:
@andrewkozlik  I would like to work on this,  since the changes will involve multiple files, I'm curious to know if I should do this in a single PR or should each file change be a single PR?

matejcik, 2024-02-29:
@naiyoma make a single PR please