trezor/trezor-firmware #2541: SLIP-25 in trezorctl get-descriptor

andrewkozlik, 2022-10-04:
In `trezorctl get-descriptor` one can't specify a [SLIP-25 account](https://github.com/satoshilabs/slips/blob/master/slip-0025.md#public-key-derivation). The user can only specify the account using script type and account number, see below. I suppose one possible solution is to redefine `-n` to mean BIP-32 path or account number. Another solution is to add a `--slip25` option to force `m / 10025' / coin_type' / account' / script_type' / change / address_index` instead of e.g. `m / 86' / coin_type' / account' / change / address_index`.

```
Usage: trezorctl btc get-descriptor [OPTIONS]

  Get descriptor of given account.

Options:
  -c, --coin TEXT
  -n, --account INTEGER           account index (0 = first account)
                                  [required]
  -t, --script-type [address|segwit|p2shsegwit|taproot|pkh|wpkh|sh-wpkh|tr]
  -d, --show-display
  --help                          Show this message and exit.
```

_Originally posted by @andrewkozlik in https://github.com/trezor/trezor-firmware/issues/2517#issuecomment-1248360130_

prusnak, 2022-10-04:
Since SLIP-0025 account has different semantics than BIP44/49/84/86, I'd prefer to have a separate option:

> Another solution is to add a `--slip25` option

prusnak, 2022-10-04:
I can also imagine two new options `--bip XX` and `--slip XX`, where
* `--bip 44` equals `--script-type address` and `--script-type pkh`
* `--bip 49` equals `--script-type segwit` and `--script-type wpkh`
* `--bip 84` equals `--script-type p2shsegwit` and `--script-type sh-wpkh`
* `--bip 86` equals `--script-type taproot` and `--script-type tr`

matejcik, 2022-10-05:
> I can also imagine two new options `--bip XX` and `--slip XX`, where

more like `-a,--account-type` so `-a bip44` `-a slip25`, it's better than two distinct `--bip/--slip` options where we would need to manually resolve "at most one of" condition

prusnak, 2022-10-05:
> more like `-a,--account-type` so `-a bip44` `-a slip25`

👍

closed 2023-03-20 by andrewkozlik