trezor/trezor-firmware #2558: each icon (for IconDialog) takes 0.5K of flash, we need to either figure out how to represent three-color image more efficiently than 16bpp, or draw them procedurally

sime, 2022-10-13:


prusnak, 2022-10-13:
> three-color image

Which images are these?

matejcik, 2022-10-13:
[image]

black background + yellow triangle + white excl mark + alpha-blends on borders
to save space, we need to compose "4bpp triangle" + "4bpp excl mark" (as opposed to rgb565 of the whole thing)

prusnak, 2022-10-13:
I think we could create a function that takes a list of tuples with icon data and combines these together?

Something like:

```
combine_icons(x, y, w, h, background_color, icons)
```

where icons is a list of tuples: `(icon_x, icon_y, icon_data, render_color)`

matejcik, 2022-10-13:
yes, that is probably the solution. we could not do this before however, because before #2471 we did not have the capability to blend icons over one another

prusnak, 2022-10-13:
> we did not have the capability to blend icons over one another

I meant the software blending. The `combine_and_render_icons` functions would go over pixels and compute each pixel color.

``` python
for yy in range (h):
   for xx in range(w):
      color = 0
      for icon in icons:
         c = compute_icon_color(icon, xx, yy)
         color += c
      color /= len(icons)
      push_pixel_data_to_display(color)
```

PR #2573 (2022-10-22): Render three color icons by blending two grayscale
Saves some flash space and is a bit faster.

Currently the `BlendedImage` component draw both icons centered, if we need something else in the future it should be easy to modify.

Uses DMA2D for blending.

When DMA2D is not available, defaults to drawing one icon first and then another, which works fine but is prone to flickering, is slower and is limited to non-overlapping borders (= only works correctly if the FG icon is rendered over only one color of the BG icon).

Closes #2558
closed 2022-10-22 by mmilata