trezor/trezor-firmware #2560: Add Proton mail profile to Trezor firmware

bosomt, 2022-10-14:
Recently Proton enabled two-factor authentication keys (2FA keys) login and Trezor works as expected.
So we can add U2F definition to our firmware.

https://proton.me/blog/security-keys

[image]

prusnak, 2022-10-14:
JFYI

```
>> hashlib.sha256(b"proton.me").hexdigest()
'a259c2b50d785080f8be7f17caf8156c8d18f47edbaf518fa6f59f29cd28f15c'
```

prusnak, 2022-10-14:
PR in https://github.com/trezor/trezor-firmware/pull/2563

bosomt, 2022-10-17:
**QA OK**

**Info:**
 - Suite version: desktop 22.10.2 (4fecea60b5c0f2777708c67640375a753bd49234)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/22.10.2 Chrome/100.0.4896.160 Electron/18.3.5 Safari/537.36
 - OS: MacIntel
 - Screen: 1440x900
 - Device: model T 2.5.3 Universal (revision 498422ce5db85a9a279f6e8c8fce659cd12ea679)
 - Transport: bridge 2.0.31

closed 2022-10-17 by bosomt