trezor/trezor-firmware #2579: Setting `auto-lock-delay` does not appear to affect the lock timeout

webstrand, 2022-10-21:
**Describe the bug**
Setting the `auto-lock-delay` to `14400` does not work, the device locks after 10 minutes.

**Firmware version and revision:**
Universal 2.5.2

**Desktop/smartphone setup:**
 - Wallet software: trezorctl
 - OS: Linux
 - Version 0.13.3

**To Reproduce**
Steps to reproduce the behavior:
1. Connect Trezor-T to computer
2. Unlock the device with pin
3. `trezorctl set auto-lock-delay 14400`
4. Confirm a timeout of "4 hours" on the device.
5. Wait 10 minutes, the device will lock.

**Expected behavior**
The device should lock around 4 hours later, after being unlocked.

**Additional context**
Recently wiped and upgraded the firmware from 2.4.3 to 2.5.2. Previously, this setting was applied correctly. I don't know which version of the firmware I had when previously set.

webstrand, 2022-10-21:
I've apparently had this _exact_ issue before: https://gitter.im/trezor/community?at=5fb5d5d37cac87158fb5b82a previous version was most likely 2.3.4 when the setting finally took.

I've additionally tried restarting the device, it still locks after 10 minutes (± 4 seconds).

bosomt, 2022-10-26:
I was not able to reproduce issue with 2.5.2 public version or 2.5.3 dev master build.
Change settings via trezorctl multiple times to 14400 or 6000 minutes but device never locked it self after 10 minutes.

matejcik, 2022-11-09:
yeah, unfortunately we can't do anything because the feature seems to work perfectly on our end :(
let us know if you figure out how to get into a scenario where this happens again, i suppose?

closed 2022-11-09 by hynek-jina