trezor/trezor-firmware #2585: Usage of newlines in firmware strings

grdddj, 2022-10-26:
As a "relict" from `UI1`, there are quite a lot of places where we are inserting newlines `\n` into a to-be-displayed string to visually separate some parts of the message.

Example, where it makes sense, is `"The mining fee of\n{}\nis unexpectedly high."` However, mostly for longer sentences, it seems to be not so useful.

Also, considering that `TR` will be using the same strings, we may probably delete majority of these.

### Current:
```python       
content = "The following\ntransaction output\ncontains tokens."
```
[image]

### Suggested:
```python
content = "The following transaction output contains tokens."
```
[image]

grdddj, 2022-10-27:
> Rather than a specific task, couldn't we do this as we go along?
> 
> i.e. Whenever you touch a particular layout, clean up the newline characters?

Also a possibility, but we can quite easily find out all the newlines in strings at once and decide what to do with them. (For example, delete all of them in one commit and see the `UI` diff.)

In regards to `TR`, it would be nice not having to do it there, as these changes would retrospectively change the `TT`design. (In the current `TR` branch, I am deleting the newlines in a special function, which is not very good - we want to probably keep some of the newlines even for `TR`.)

mmilata, 2022-10-31:
UI2 works around this in couple of places like
```
title=content.replace("\n", " ")
```
we should get rid of these after we switch.

closed 2023-02-07 by grdddj