trezor/trezor-firmware #2598: Not displaying the passphrase by default

Hannsek, 2022-11-02:
Introduce settings in `trezorctl` that would disable displaying passphrase on device by default.

User will have a choice to view the passphrase or not.
[image]

The default behaviour stays unmodified.

prusnak, 2022-11-02:
I don't think that's we want. The default should be to show the passphrase. Not showing the passphrase should be an opt-out.

Hannsek, 2022-11-02:
Default will be same as now -> passphrase visible by default

This feature is opt-in through trezorctl.

prusnak, 2022-11-02:
Ah okay. But in that case the feature doesn't make sense if there is a `View Passphrase` button.

The whole point of this feature was to prevent the passphrase from showing on the device.

sime, 2022-11-02:
T1?

Hannsek, 2022-11-03:
It does make sense to me. I imagine the situation when you want to access your wallet in front of somebody and don't want to show them the passphrase. They won't see it and you are still able to check what you wrote on the device. 

@prusnak What you are proposing is to not showing passphrase at all so you can never check what you wrote.

I don't see a reason why not to include this `View Passphrase` button.

prusnak, 2022-11-03:
You are missing the usecase why this feature was requested in the first place. Let's talk on Slack

andrewkozlik, 2022-11-03:
I don't understand the proposal in the OP - what are the two parameters?

On T1 we can implement the "View Passphrase" button in two consecutive screens:
Access hidden wallet? [No] [Yes]
View passphrase? [Skip] [View]

Some things to think about:
1. Checking the passphrase is important if you want to do GetAddress, but less important if you want to do SignTx (assuming you have only one passphrase, or one real and one decoy passphrase). So the View passphrase button seems like it should be the default workflow.
2. Do we need to distinguish between show passphrase always and show passphrase behind "View Passphrase" button? Let's think about forgetting the first option altogether.
3. Do we need the "Access hidden wallet?" screen if we are not showing the passphrase at all? I think yes, although I think we didn't have it before this feature was introduced.
4. I have some doubts about the whole user-not-present scenario. If there is someone else present, then they can sniff the USB communication for the passphrase anyway.

For reference this feature was introduced in: cdf0f68bb03208faf2d7050aa1706635522467c3

andrewkozlik, 2022-11-04:
My opinions:
1. Come to think of it (we probably had this discussion in 2020 already), the option which allows the user to decide on device whether to view the passphrase or not, can already be implemented physically by the user. If there is someone present, then they just cover the screen and click next.
2. Either forget about the show-passphrase-always or the view-passphrase-button, it's just seems like additional complexity to have both. I'd just stick with the current workflow to keep it simple.
3. The reason why I think yes is because otherwise an attacker who controls the host computer can switch to a different passphrase without the user knowing. For example the user thinks they are not using a passphrase at all, but the attacker sets a passphrase. The user does `GetAddress` and then sends a large amount to the address. Then the user cannot recover the funds without the attacker's assistance. The user should at least know that some passphrase is being used and then they can use other means to verify that it is the passphrase they expect, e.g. doing `GetAddress` for an already used path.
4. The user-not-present scenario is a concept that doesn't fit into our security model. Not only can the passphrase be sniffed but the session can be hijacked by whoever is present. Both of these problems can be mitigated by https://github.com/trezor/trezor-firmware/issues/79. But there are other problems still, e.g. the present party can take over control of the Trezor once it's unlocked. So they won't have control of the passphrased session, but they will have access to the unlocked Trezor. That can be solved by unlocking the device only for a particular session as opposed to all sessions like we do now. Finally there is the problem of physical attacks.

Hannsek, 2022-11-07:
1. Behaviour of showing passphrase behind "View passphrase" button was intended to prevent _side-channel attack_ on model R.
2. Having both option can be beneficial as the users would choose which behaviour of Trezor they want.
3. Agree
4. The behaviour of not showing passphrase at all can be useful for very specific situation(s). -> @prusnak

matejcik, 2022-11-07:
in any case, any behavior different from the current one should be gated behind safety checks.

Hannsek, 2022-11-07:
Final decision is that we only introduce the ability to not showing the passphrase at all.

prusnak, 2022-12-11:
I thought this was also scheduled to be done at the same time as https://github.com/trezor/trezor-firmware/issues/450 

Can we please make this happen and fit this into the next release? The change seems trivial enough now that we have reduced the scope.

Hannsek, 2022-12-12:
It will be implemented until next release which will be in February.

Hannsek, 2022-12-12:
@matejcik https://github.com/trezor/trezor-firmware/pull/2675

prusnak, 2022-12-12:
> draft content: "⚠️ Passphrase sent from host will not be displayed. [proceed]"

Nice, OK

> also, looks like we will not need this for T1 after all, correct?

Right

prusnak, 2022-12-12:
> Do we have to show any screen at all?

No strong requirement from my side, feel free to decide whether we want to show the screen

matejcik, 2022-12-12:
> Do we have to show any screen at all?

We don't _have to_ but we _should_. Using this feature is inherently risky.

Hannsek, 2022-12-12:
It is, but it will be under _trezorctl command_, so it is up to the users.

matejcik, 2022-12-12:
plus, at the very least, it indicates that a passphrase is being used at all, which just seems like the nice thing to do?

prusnak, 2022-12-12:
Maybe we can use less drastic copy (without the warning sign) and just say:

`Passphrase provided by host will be used but will not be displayed due to the device settings`?

Hannsek, 2022-12-12:
That is the thing. Not sure if this is something you want to show to someone, that you are using passphrase…

prusnak, 2023-02-06:
Let's make also sure that the signing tools check for the setting in the Features flag. We want to make sure the devices used for signing are configured to NOT show the passphrase.

closed 2023-02-17 by grdddj