trezor/trezor-firmware #2602: Compress bootloader to reduce size of the firmware

prusnak, 2022-11-03:
``` python
import zlib
bl = open("bootloader.bin", "rb").read()
z = zlib.compressobj(level=9, wbits=-10)
zp = z.compress(bl) + z.flush()
>>> len(bl), len(zp), len(bl) - len(zp)
(94208, 61366, 32842)
```

We don't need extra RAM, we can just write to flash on the fly, check the hash of the written bootloader and repeat if something went bad. Repeat this 3 times, then throw an error.

TychoVrahe, 2022-11-04:
my worry was that flashing while decopressing increases the chance of something going wrong and thus increases the chance of bricking the device. On the other hand this should still be deterministic process and if tested enough should be ok - a bit slower but hopefully not enough to give user time to disconnect the device

prusnak, 2022-11-04:
Yeah, we should test what is the reliability of this method first (maybe by creating a special firmware which expands the bootloader to another sector - not the bootloader sectors) and also testing how worse is the UX (i.e. how much longer does an user have to wait).

matejcik, 2023-11-28:
implemented in #3304

closed 2023-11-28 by matejcik