trezor/trezor-firmware #2605: TR bootloader design

Hannsek, 2022-11-04:
Design: https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Trezor-T%2C-Trezor-One?node-id=3790%3A11545

TychoVrahe, 2022-11-21:
Some feedback:
from firmware perspective, it would be beneficial to have the same flow as in model T - keeps things nicely united, less error prone and easier to maintain. This means either removing the initial bootloader warning on TR or adding it in TT, and then have screen with enter menu/ install fw actions like this:
[image]
(this would also require addition icon design for exit menu item)

Confirming menu actions: we dont use hold to confirm on TT, no need to do it here (its unnecessary complication). Reboot action does not have to be confirmed at all.

See comparison of progress bar/spinner. The spinner is nice and the transition in success screen is smooth, but the beginning with no circle around the icon is weird.
[image]
[image]

The mono font is problematic at some screens, see (mainly the phrase 'UNSAFE, DO NOT USE!' doesn't fit one line:
[image]
[image]

Hannsek, 2022-11-21:
Noted, new design in progress.

Hannsek, 2022-11-22:
Design in original comment updated.
@TychoVrahe Please do two versions of bootloader with different types of font (mono, normal) so that we can compare them.

Hannsek, 2022-12-13:
[image]

Hannsek, 2022-12-14:
Two icons of _trash_ and _exit_ uploaded:
[image]
[image]

Hannsek, 2023-02-15:
rsod screens + wipe code screen: 
[image]
[image]

closed 2023-06-06 by grdddj