trezor/trezor-firmware #2607: Document reset-device in --help menu

sime, 2022-11-04:
`trezorctl` allows users to choose seed strength, though it's not documented. e.g.

```
trezorctl reset-device -t 256
```

grdddj, 2022-11-12:
The reason some of the commands "work but are not documented" is that these are "old API commands" that have been deprecated, but we still maintain backwards compatibility.

List of all other aliased commands is here:

(How to read it - `trezorctl reset-device` is the same as `trezorctl setup device` etc.)

https://github.com/trezor/trezor-firmware/blob/a393267461a669478393c1be1842d3be91806658/python/src/trezorlib/cli/trezorctl.py#L61-L89

Documenting these is probably not desirable, because we want people to use the "new" API. Also, it is even possible to write `trezorctl reset-device --help` and it will show all the possible argument/options, including the `-t`.

```
$ tr reset-device --help
Usage: trezorctl reset-device [OPTIONS]

  Perform device setup and generate new seed.

Options:
  -e, --show-entropy
  -t, --strength [128|192|256]
  -r, --passphrase-protection
  -p, --pin-protection
  -l, --label TEXT
  -u, --u2f-counter INTEGER
  -s, --skip-backup
  -n, --no-backup
  -b, --backup-type [single|shamir|advanced]
  --help                          Show this message and exit.
```

matejcik, 2022-11-14:
I understand this issue to be about the "perform device setup and generate new seed" text?

Like, instead of a single sentence it should be a paragraph or two, with explicit mention that `-t 128` corresponds to 12 word seed single or 20 word shares Shamir, etc.

sime, 2022-11-14:
What is the new API command for `trezorctl reset-device -t 256` ?

matejcik, 2022-11-14:
`trezorctl device setup [your same options as before go here`

sime, 2022-11-14:
Since `reset-device` options are available under `device setup` then is makes sense that it not longer documented.

closed 2022-11-14 by sime