trezor/trezor-firmware #2612: Remove coordinator identifier screen

hynek-jina, 2022-11-07:
In current version we are displaying confusing name `CoinjoinCoordinatorIdentifier`.
We could hack it, but more clean solution is to remove this screen entirely. At least as long as we only have one coordinator...

[image]

andrewkozlik, 2022-11-07:
Internal issue: https://www.notion.so/satoshilabs/Do-not-confirm-coordinator-identifier-on-Trezor-f315bc707be4459e958c0f73ab2de6ca

closed 2022-12-07 by andrewkozlik

bosomt, 2022-12-13:
**QA OK**

**Info:**
 - Suite version: desktop 22.12.1 (ddaacdb2245d7a909b3d2abfe918cff34d2e1609)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/22.12.1 Chrome/104.0.5112.124 Electron/20.3.5 Safari/537.36
 - OS: MacIntel
 - Screen: 1440x900
 - Device: model T 2.5.4 Universal (revision ff35c77bad82c7acac1df6245940ebe0d6454642)
 - Transport: bridge 2.0.32