trezor/trezor-firmware #2615: Missing dependencies from `shell.nix`?

danielabrozzoni, 2022-11-07:
**Describe the bug**
Hey, I think that the current `shell.nix` is missing a few dependencies, as I get 
```
Traceback (most recent call last):
  File "/home/daniela/Developer/bitcoindevkit/emulators/trezor-firmware/core/./tools/../vendor/trezor-common/tools/cointool.py", line 15, in <module>
    import click
ModuleNotFoundError: No module named 'click'
make: *** [Makefile:129: templates] Error 123
```
when trying to compile the emulator with `make build_unix`. Am I doing something wrong?

**Firmware version and revision**
I'm compiling in master, my latest commit is 4e800b6ffde97862026c6196f7d983d62a05ea71

**Desktop/smartphone setup (please complete the following information):**
 - OS: NixOS 22.05, build ID "22.05.3977.f09ad462c5a"

**To Reproduce**
From the root of the project:
```
nix-shell
cd core
poetry shell # I'm not sure if I should run it or not, but the problem persists if I do
make build_unix
```

matejcik, 2022-11-07:
`poetry shell` is necessary

however, you most likely skipped the initial `poetry install` step. just run that once (or every time you pull the repo) and you should be good to go

danielabrozzoni, 2022-11-07:
Ouch, right! Thanks a lot :)