trezor/trezor-firmware #2621: Add bootloader build to CI

TychoVrahe, 2022-11-09:
Currently we don't know when some PR breaks bootloader build. Going forward, checking this will be more important as bootloader will share some Rust UI code with firmware.

mmilata, 2022-11-09:
It's already there? Unless this issue is regarding TR. https://github.com/trezor/trezor-firmware/blob/master/ci/build.yml#L26

TychoVrahe, 2022-11-09:
right. missed that. so the CI check just doesn't work properly, since it didn't detect that the bootloader build is broken

mmilata, 2022-11-09:
Looks like `core/tools/headertool.py` returns 0 exit code regardless of signature/hashes.

matejcik, 2022-11-10:
Properly parsed file, regardless of signature/hashes status, is a success for headertool so that part is right though?

There are two parts to #2616:
(a) headertool is supposed to write correct hashes into the header, which it didn't do because as part of the refactor having the hashes is now not a prerequisite for signing. this is a bug that needs to be fixed
(b) headertool currently does not recognize DEVEL signatures for bootloader, which is more of a missing feature that needs to be re-added
neither of which make sense to catch via CI, unless we have an independent tool that verifies the result of headertool

This suggests to me that the CI part is in fact correct and we can close this issue?

TychoVrahe, 2022-11-10:
Ok sounds good enough

closed 2022-11-10 by TychoVrahe