trezor/trezor-firmware #2623: Add model information to firmware headers

matejcik, 2022-11-09:
trezor-core header should have a **single character** field `model` to indicate for which model the object is intended.

A good space for this is in the 8 reserved bytes after `fix_version`.

* legacy header should have model `'1'`. Legacy bootloader should check this field and refuse to install firmware if it's not there
  * ideally as part of #2513 
* core bootloader header should contain the field (set to 'R' or 'T'). There is no requirement for the boardloader to check this though; even if a mismatch is found, there is nothing the boardloader or the user can do about it. Just try to load it and hope for the best.
* core firmware header should contain the field (set to `'R'` or `'T'`). The bootloader must check this field and refuse to install firmware if there is a mismatch.
  * TT bootloader may want to accept older firmwares where this is set to `0`.
* for the hypothetical R/T multifirmware, we might choose a specific character (`'m'` for multi, perhaps?) or allow more characters

open question: should vendor header also be model-specific?

TychoVrahe, 2022-11-09:
Current TT bootloaders currently don't check for the model version. So when we release model R, the non-upgraded bootloaders will be still able to flash and run the TR firmware and potentially brick the device. 

To ensure this doesn't happen we could:
- check board-capabilities when installing bootloader whether it is compatible
- and/or modify image header format so that new TR fw images are evaluated as invalid in older bootloader, but new TT images are still valid somehow. Preventing the incompatible fw to run would be better - who knows what can happen to the incompatible hw, apart from installing bad bootloader. 

Also, a runtime check when starting firmware in addition to flash-time check could be useful.

Vendor header- at minimum these contain vendor images, which are somewhat model specific (model R can't show the current ones). Since these not shown currently it might not be that important though

matejcik, 2022-11-10:
we could bump `expiry` to 1 to make the header invalid for existing TT bootloaders.

from a very quick thought, this should not pose a forward compatibility problem; if we ever implement an actual expiry mechanism, the value 1 would mean "expired long ago" which is the correct thing.

Hannsek, 2022-12-09:
How about 
`< _generation_ >.T1` for entry level models
`< _generation_ >.TT` for middle level models
`< _generation_ >.T3` for high end level models

For example:
1.T1 - Trezor model One
2.T1 - Trezor model R
1.TT - Trezor model T
2.TT - New Trezor model T

TychoVrahe, 2022-12-09:
Since we are limited by characters i would drop anything that does not add value - here '.' and first 'T'. Or maybe we could keep the 'T' and differentiate entire classes of devices, in case we at some produce something not a Trezor, like some accessories or something. 

So something like
T101 - Trezor model One 
T102 - Trezor model R 
TT01 - Trezor model T 
N101 - Bitcoin Full Node model One, for example..
etc.

matejcik, 2022-12-09:
that is reasonable, so we have four bytes:
* `class` - `'T'` for Trezor hardware wallet
* `generation` - we can go with ascii, when we're on gen 10 we can continue with letters
* `model` - this can be whatever, one or possibly two bytes

for consideration: product-wise TT is a "first generation" device and TR is a "second generation", but code-wise the only first gen product is T1, while both TT and TR are second gen. A "new TT" will most likely be 3rd gen, but could in theory also end up as 2nd gen if we go with the same platform as TR/TT.

That gives me two options.
One, slightly more preferable for us:
`T11 ` for Trezor One
`T2T ` for Trezor T
`T2R ` or `T21 ` for Trezor R
`T3T ` for "new TT"

Other, friendlier for Product possibly?
`T11 ` for Trezor One
`T1T ` for Trezor T
`T21 ` for Trezor R
`T2T ` for "new TT"

matejcik, 2022-12-09:
a little more refinement still:
* `product_class` - `'T'` for Trezor hardware wallet
* `platform` - `'1'` for STM32F207, `'2'` for STM32F42x
* `feature_class` - `'B'` for Buttons, `'T'` for Touch... (`'w'` for wireless, `'g'` for gold plated, `'h'` for holographic, `'K'` for keyfob (blind device with just a button) etc.)
* `reserved` - reserved for future use

prusnak, 2022-12-09:
> a little more refinement still:

I like this proposal the most!

TychoVrahe, 2022-12-09:
i think we still gonna need the generation identifier even if we add platform

prusnak, 2022-12-09:
> generation identifier

But what is a generation? How is it defined?

TychoVrahe, 2022-12-09:
> start using reserved as generation

that would be my favorite. 

where generation is basically just a number assigned when development starts (or product decision is made). 

a question is whether to number in specific feature class or globally, but specific feature class seems a bit better, we can fit in more information

TychoVrahe, 2022-12-09:
so in this scheme it would be like:
T1B1 - Model One
T2T1 - Model T
T2B2 - Model R. or even T2B1?
R2D2 - something else entirely :)

Hannsek, 2022-12-09:
@TychoVrahe T2B1 as "model R" will be the first (1) Trezor (T) with buttons (B) on F42x (2).

This is fine. My concern is that it would be more prone to error. But I think we can handle it.

matejcik, 2022-12-10:
> since we can have several trezor devices with touch screen running on F42x etc

I meant the "feature class" in a more general sense. Yes, we can have several Trezors on the same F42x that all have a touch screen, but no two should be in the same feature class, otherwise it makes little sense to have them as separate products?
Like, one might be fully airgapped in which case the feature class is "A"

But given the following feature matrix on the same platform:
* buttons + USB
* buttons + airgap
* touch + USB
* touch + airgap

Then it does not make sense to differentiate them by generation, _and_ we can't fit the features in a single ASCII "feature class".
So what would make sense to me is make the two last bytes "model" + "differentiator", so the above matrix would be: `T2B\0` / `T2Ba` / `T2T\0` / `T2Ta`. Which also maps to keeping the "differentiator" as `reserved` for now.

matejcik, 2022-12-10:
and we can add "generation" instead of "device class", and differentiate device class by the magic bytes, which might make more sense anyway

TychoVrahe, 2022-12-10:
Even with the more general feature class definition there can easily be case where we would have two different products with the same features.

Like model T facelift, which would include lets say include updated design and 16 bit bus for the display to improve speed, which would make it software-wise incompatible. Unless you include the tech spec into the feature class, but then it is no more feature class.

Hannsek, 2022-12-12:
I would go with @matejcik's proposal: https://github.com/trezor/trezor-firmware/issues/2623#issuecomment-1344243581

< `product_class` > < `platform` > < `feature_class `> < `generation` >

where `feature_class` will be something very distinct for the one particular device

PR #2649 (2023-01-12): Prevent installing and running incompatible firmware
Aims to disable running incompatible firmware to prevent
- installing incompatible bootloader and thus possibly bricking the device
- damaging HW by potentially setting the pins to something incompatible

Additionally, when installing bootloader, the bootloader monotonic version is checked to prevent downgrade, which would also cause bricking the device. 

Added 3 items to image header
- monotonic version
- hw model
- hw revision

Bootloader checks fw's image header to ensure compatible hw model and revision - both when installing and runnin [... cut]
closed 2023-01-12 by TychoVrahe