trezor/trezor-firmware #2625: Reproducibility of Legacy builds across x86_64 and arm64 platforms

prusnak, 2022-11-10:


prusnak, 2022-11-10:
With Legacy the main issue is that:
a) newlib maintainers leak build paths to their builds (each distribution is built differently and builds for different archs have different paths)
b) we leak this paths from newlib to our firmware binaries

This can be confirmed easily:

```
$ xxd trezor-1.11.2.bin | grep jenkins -A 4 -B 4
000b4980: 0000 0000 4261 6c6c 6f63 2073 7563 6365  ....Balloc succe
000b4990: 6564 6564 0000 0000 2f6d 6e74 2f77 6f72  eded..../mnt/wor
000b49a0: 6b73 7061 6365 2f77 6f72 6b73 7061 6365  kspace/workspace
000b49b0: 2f47 4343 2d31 302d 7069 7065 6c69 6e65  /GCC-10-pipeline
000b49c0: 2f6a 656e 6b69 6e73 2d47 4343 2d31 302d  /jenkins-GCC-10-
000b49d0: 7069 7065 6c69 6e65 2d33 3338 5f32 3032  pipeline-338_202
000b49e0: 3131 3031 385f 3136 3334 3531 3632 3033  11018_1634516203
000b49f0: 2f73 7263 2f6e 6577 6c69 622f 6e65 776c  /src/newlib/newl
000b4a00: 6962 2f6c 6962 632f 7374 646c 6962 2f64  ib/libc/stdlib/d
000b4a10: 746f 612e 6300 0000 2f6d 6e74 2f77 6f72  toa.c.../mnt/wor
000b4a20: 6b73 7061 6365 2f77 6f72 6b73 7061 6365  kspace/workspace
000b4a30: 2f47 4343 2d31 302d 7069 7065 6c69 6e65  /GCC-10-pipeline
000b4a40: 2f6a 656e 6b69 6e73 2d47 4343 2d31 302d  /jenkins-GCC-10-
000b4a50: 7069 7065 6c69 6e65 2d33 3338 5f32 3032  pipeline-338_202
000b4a60: 3131 3031 385f 3136 3334 3531 3632 3033  11018_1634516203
000b4a70: 2f73 7263 2f6e 6577 6c69 622f 6e65 776c  /src/newlib/newl
000b4a80: 6962 2f6c 6962 632f 7374 646c 6962 2f6d  ib/libc/stdlib/m
```

I am not even trying to provide diff for legacy builds, because the jenkins paths from newlib have different lengths resulting in completely unusable diffs.

prusnak, 2022-11-10:
I updated gcc-arm-embedded locally to version 11 and the issue still persists with FW built from master using gcc-arm-embedded-11:

```
$ strings legacy/firmware/firmware.bin | grep -i jenkins
/data/jenkins/workspace/GNU-toolchain/arm-11-2/src/newlib-cygwin/newlib/libc/stdlib/dtoa.c
/data/jenkins/workspace/GNU-toolchain/arm-11-2/src/newlib-cygwin/newlib/libc/stdlib/mprec.c

$ xxd legacy/firmware/firmware.bin | grep jenkin -A 10 -B 10
000b5130: 4152 5900 4c43 5f4e 554d 4552 4943 0000  ARY.LC_NUMERIC..
000b5140: 4c43 5f54 494d 4500 4c43 5f4d 4553 5341  LC_TIME.LC_MESSA
000b5150: 4745 5300 496e 6669 6e69 7479 0000 0000  GES.Infinity....
000b5160: 4e61 4e00 4261 6c6c 6f63 2073 7563 6365  NaN.Balloc succe
000b5170: 6564 6564 0000 0000 2f64 6174 612f 6a65  eded..../data/je
000b5180: 6e6b 696e 732f 776f 726b 7370 6163 652f  nkins/workspace/
000b5190: 474e 552d 746f 6f6c 6368 6169 6e2f 6172  GNU-toolchain/ar
000b51a0: 6d2d 3131 2d32 2f73 7263 2f6e 6577 6c69  m-11-2/src/newli
000b51b0: 622d 6379 6777 696e 2f6e 6577 6c69 622f  b-cygwin/newlib/
000b51c0: 6c69 6263 2f73 7464 6c69 622f 6474 6f61  libc/stdlib/dtoa
000b51d0: 2e63 0000 2f64 6174 612f 6a65 6e6b 696e  .c../data/jenkin
000b51e0: 732f 776f 726b 7370 6163 652f 474e 552d  s/workspace/GNU-
000b51f0: 746f 6f6c 6368 6169 6e2f 6172 6d2d 3131  toolchain/arm-11
000b5200: 2d32 2f73 7263 2f6e 6577 6c69 622d 6379  -2/src/newlib-cy
000b5210: 6777 696e 2f6e 6577 6c69 622f 6c69 6263  gwin/newlib/libc
000b5220: 2f73 7464 6c69 622f 6d70 7265 632e 6300  /stdlib/mprec.c.
000b5230: 0080 e037 79c3 4143 176e 05b5 b5b8 9346  ...7y.AC.n.....F
000b5240: f5f9 3fe9 034f 384d 321d 30f9 4877 825a  ..?..O8M2.0.Hw.Z
000b5250: 3cbf 737f dd4f 1575 0000 0000 0000 f03f  <.s..O.u.......?
000b5260: 0000 0000 0000 2440 0000 0000 0000 5940  ......$@......Y@
000b5270: 0000 0000 0040 8f40 0000 0000 0088 c340  .....@.@.......@
```

matejcik, 2022-11-11:
aside: a very random dive into the binary reveals that we carry, e.g., Japanese locale support, with newlib.
probably not worth the effort now, but it would have been better to use a more pared down version of a standard library

matejcik, 2022-11-11:
stupid idea that will probably work:
just take `libc.a` from the x86_64 build and place it into the arm64 toolchain

prusnak, 2022-11-11:
I am trying to figure out what exactly requires newlib by passing `-nostdlib` arg to CFLAGS and LDFLAGS:

``` patch
diff --git a/legacy/Makefile.include b/legacy/Makefile.include
index b5149c4fe..5d65ebf78 100644
--- a/legacy/Makefile.include
+++ b/legacy/Makefile.include
@@ -50,6 +50,7 @@ CFLAGS   += $(OPTFLAGS) \
             $(DBGFLAGS) \
             $(SANFLAGS) \
             -std=gnu11 \
+            -nostdlib \
             -W \
             -Wall \
             -Wextra \
@@ -88,6 +89,7 @@ CFLAGS   += $(OPTFLAGS) \
             -I$(TOP_DIR)vendor/trezor-storage
 
 LDFLAGS  += -L$(TOP_DIR) \
+            -nostdlib \
             $(DBGFLAGS) \
             $(SANFLAGS) \
             $(CPUFLAGS) \
```

Oddly enough, the build does not change at all. 🤷

prusnak, 2022-11-11:
I tried to build with the following patch to see which functions are required from newlib:

``` patch
diff --git a/legacy/Makefile.include b/legacy/Makefile.include
index b5149c4fe..658ee5727 100644
--- a/legacy/Makefile.include
+++ b/legacy/Makefile.include
@@ -50,6 +50,7 @@ CFLAGS   += $(OPTFLAGS) \
             $(DBGFLAGS) \
             $(SANFLAGS) \
             -std=gnu11 \
+            -nostdlib \
             -W \
             -Wall \
             -Wextra \
@@ -122,13 +123,12 @@ CFLAGS   += -DRAND_PLATFORM_INDEPENDENT=1
 
 LDFLAGS  += --static \
             -Wl,--start-group \
-            -lc \
-            -lgcc \
             -lnosys \
             -Wl,--end-group \
             -L$(TOOLCHAIN_DIR)/lib \
             -T$(LDSCRIPT) \
             -nostartfiles \
+            -nostdlib \
             -Wl,--gc-sections
 
 LDLIBS   += -ltrezor
```

It's these functions:

```
__aeabi_uldivmod
_ctype_
explicit_bzero
gmtime
memcmp
memcpy
memmove
memset
snprintf
strchr
strcmp
strcpy
strftime
strlcat
strlcpy
strlen
strncmp
strnlen
```

Most of these are obvious, less obvious are `__aeabi_uldivmod` and `_ctype_ `

The first is used every time we perform a division of `uint64_t`:

```
legacy/firmware/layout2.c:510: undefined reference to `__aeabi_uldivmod'
legacy/firmware/signing.c:1938: undefined reference to `__aeabi_uldivmod'
legacy/firmware/stellar.c:1419: undefined reference to `__aeabi_uldivmod'
```

The latter is used when we call `toupper()` from `<ctype.h>`:

```
legacy/firmware/layout2.c:993: undefined reference to `_ctype_'
legacy/firmware/layout2.c:1034: undefined reference to `_ctype_'
legacy/firmware/recovery.c:268: undefined reference to `_ctype_'
```

prusnak, 2025-04-30:
btw I just noticed that we use newlib for legacy an newlib-nano for core

the following patch switches to newlib-nano for legacy too:

``` patch
 --- a/legacy/Makefile.include
+++ b/legacy/Makefile.include
@@ -131,7 +131,7 @@ CFLAGS   += -DEMULATOR=0
 
 LDFLAGS  += --static \
             -Wl,--start-group \
-            -lc \
+            -lc_nano \
             -lgcc \
             -lnosys \
             -Wl,--end-group \
```

but the issue still persist even on newlib-nano:

```
$ strings build/legacy-T1B1/firmware/firmware.bin | grep -i jenkins
/data/jenkins/workspace/GNU-toolchain/arm-12/src/newlib-cygwin/newlib/libc/stdlib/dtoa.c
/data/jenkins/workspace/GNU-toolchain/arm-12/src/newlib-cygwin/newlib/libc/stdlib/mprec.c
```

prusnak, 2025-04-30:
the issue is still present when using gcc-arm-embedded-14:

[image]