trezor/trezor-firmware #2627: TR design - Seed backup

Hannsek, 2022-11-11:
[image]

grdddj, 2022-11-14:
Done in [b723445](https://github.com/trezor/trezor-firmware/pull/2610/commits/b723445b6b61d582aafbbedc44a9ec038597844f)

[image]

Hannsek, 2022-11-14:
007 - text too low
009 - `Confirm` instead of `Hold to begin`
013, 015 - these screen are not supposed to be here.
017 - unnecessary as 019 says the same.
`Backup is done` screen is missing
085 - `Check phrase` instead of `Check backup`, text shouldn't be **bold**
087-096 - `Select word X/Y` instead of `Select X word` _(we don't have to change that)_
099 - duplicate of 019

grdddj, 2022-11-14:
> 007 - text too low
> 009 - Confirm instead of Hold to begin

These screens are shared with `recovery`, so the button on the right side will be the same in both cases.

> 013, 015 - these screen are not supposed to be here.

This would however mean that users would not be able to skip backup. It is a good thing, but it is changing the flow we have. Do we really want to enforce the backup?

> 017 - unnecessary as 019 says the same.

17 is what we do generally right now, 19 is specific for `model R`. Do we want to skip `17` therefore?

> Backup is done screen is missing

> 087-096 - Select word X/Y instead of Select X word (we don't have to change that)

The "st", "nd", "rd", "th" can be done, we already do it somewhere. I see I am also not centering that, will do. 

> 099 - duplicate of 019

That is because I have entered a wrong word and therefore it started from the beginning. Is it alright?

Hannsek, 2022-11-14:
User would firstly _create_ new wallet and then back it up with the seed. Different header than in `recovery` flow.
-> changed first 2 screens and added one more screen with `skip backup`.

Drop `017`.

099 is alright.

Hannsek, 2022-12-19:
New design/copy/screens: 
[image]

closed 2023-05-31 by grdddj