trezor/trezor-firmware #2628: TR design - Recovery

Hannsek, 2022-11-11:
**# of words**
[image]
**Word entry**
[image]

grdddj, 2022-11-14:
Ready to review from https://github.com/trezor/trezor-firmware/pull/2610:

[image]

The screen `START RECOVERY: Select number of words` was already part of our logic, so I kept it instead of extending the `TOS` sequence. We might extend the sentence and therefore show `First select the number of words in your recovery seed` on `model T` as well.

Hannsek, 2022-11-14:
We want to extend the sentence as users were a bit confused with `Select number of words`.

Hannsek, 2022-11-14:
071 - text too low, `Confirm` instead of right arrow, left arrow not in _button_ style
075 - text (see above), `Start` instead of `Hold to begin`
083 - IMO unnecessary screen
087-126 - header `word X of Y` not centered

grdddj, 2022-11-14:
> 071 - text too low, Confirm instead of right arrow, left arrow not in button style
> 075 - text (see above), Start instead of Hold to begin
> 083 - IMO unnecessary screen
> 087-126 - header word X of Y not centered

Addressed `71-a`, it is currently centered and would need some nontrivial changes to be moved higher.

`71-b,c` - that comes from the fact that it is not possible to put 69, 71 and 75 into the same sequence, these are called from different places.

Addressed `75` - as mentioned, this screen is special, and it comes again even when you disconnect your Trezor - it remembers you already confirmed you want to do recovery. Added `HOLD TO BEGIN`, but as you can see below, it also put the same `HOLD TO CONFIRM` in `69`, as they are the same function/layout. We might put different words there in different situations, but with some effort.

`83` - as above, this screen is really mandatory to have. When you disconnect Trezor during word recovery, this will appear again after you connect it.

Addressed `87-126` - centering the header

[image]

Hannsek, 2022-11-14:
071 (007) - changed design in original post (`Continue` instead of `Confirm`)
075 (011) - slightly different copy, header
083 (069) - different copy, header

Hannsek, 2022-12-19:
[image]

[image]