trezor/trezor-firmware #2629: Improve UI2 debuglink output

matejcik, 2022-11-11:
The current format looks like "< Title: bla bla content: ble ble < klsfsf >>"
It's a very rough "markup" like thing that's difficult to process in an automated manner.
It seems relatively easy to produce JSON instead.

mmilata, 2022-11-14:
Related: `_Rustlayout.read_content` currently allocates a lot of small strings and then concatenates them. Would be nice if python received just the final result(s).

matejcik, 2022-11-14:
yeah but I don't think that can be done without moving the very same allocation to Rust.

except maybe if we allocated a big vstr for a work buffer and then passed it to Python as a whole

mmilata, 2022-12-11:
json sounds good, tests can query it with https://github.com/mwilliamson/jq.py