trezor/trezor-firmware #2630: TR design - Scrollbar

Hannsek, 2022-11-11:
3 screens | 4 screens | 5+ screens
[image]

grdddj, 2022-11-14:
Done in [02ec1ac](https://github.com/trezor/trezor-firmware/pull/2610/commits/02ec1ac0b58838c9367b52bb01588bb5a55bcd4c) and [0f01a48](https://github.com/trezor/trezor-firmware/pull/2610/commits/0f01a48ba89f235edb5f9d5791a3e7b422b558b5).

Placing scrollbar at the title line makes longer titles impossible to render completely, see some screens from https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/3315073924/artifacts/test_ui_report/all_unique_screens.html

It could be solved by choosing a shorter title or showing longer titles via animation in the smaller region

Hannsek, 2022-11-14:
I would solve it with the animation. See (on the right): https://www.figma.com/proto/l0gG9XeRJ8FTDQ3cfb1wv9/Trezor-T%2C-Trezor-One?page-id=3790%3A11545&node-id=5141%3A26329&viewport=11037%2C4757%2C1&scaling=min-zoom&starting-point-node-id=5141%3A26329

closed 2023-05-31 by grdddj