trezor/trezor-firmware #2631: UI2: Implement Ethereum pagination actions

grdddj, 2022-11-11:
In https://github.com/trezor/trezor-firmware/pull/2580 two `Ethereum` device tests are skipped by `pytest.skip`, because their logic is not implemented in `UI2`.

These are tests from `test_signtx_data_pagination` test-case and they deal with longer paginated data and possibility to skip them (`input_flow_skip` ) and dive into them and go back (`input_flow_go_back`). 

Theoretically, this functionality could be brought by https://github.com/trezor/trezor-firmware/pull/2569

mmilata, 2022-12-05:
This is indeed fixed in #2569.

closed 2022-12-05 by mmilata

bosomt, 2022-12-17:
**QA OK**

 - Device: model T 2.5.4 Universal (revision 305b7fe84d802db7f0826d07a222a886f73732f0)

tested via:

`
trezorctl ethereum sign-message AAAAAgAAAADyaSwK0riaWTLCeObnbPy3K4f0n5O8g7cCiuCuPVV50wAAAGQBk/eCAAAABAAAAAEAAAAAAAAAAAAAAABht5zRAAAAAQAAABM1MTkzYjE0NWI0ODE0NzllYTUxAAAAAAEAAAAAAAAAAQAAAAB9h5rdSVjH6gHVJQ3slLTvVfMz6idGixvCq2cl7+/EBgAAAAAAAAAAAJiWgAAAAAAAAAABPVV50wAAAEDY3pOkgh840BbJsfJAp+cye66CSNCM7Cy8aPCaK/QRJ9CoEsjlU22m9A9HoymPI7vqOMr77lamP2H/dsgdgasdsaudaskhjdhksjadghkasdgasghdasjhgdjsaghdgjasdgjhasgjdsagjdasghjdgjhas73xKbfoE -n "m/44'/60'/0'/0/0"
`