trezor/trezor-firmware #2634: Differentiate trezorui2 according to model - TT, TR, etc.

grdddj, 2022-11-12:
As suggested in https://github.com/trezor/trezor-firmware/pull/2538#discussion_r1020196138, it would be nice not to put all the `Rust UI API` code in the same module - `trezorui2` (`trezorui2.pyi` file on the `upy` side).

Just the duplication of functions there (like `confirm_action`) brings confusion. Also some of the functions can take different arguments (despite efforts like https://github.com/trezor/trezor-firmware/issues/2440), just because of different screen sizes/capabilities of both models. The latter causes `pyright` issues, and it would be quite unclean to somehow silence them through `type: ignore`.

Possibility:
- we already have the `trezorui` module, we could have `trezorui_tt` and `trezorui_tr` with exposing specific functions for a specific hardware model
  - `trezorui_tt` would be only used in `core/src/trezor/ui/layouts/tt_v2` and `trezorui_tr` would be only used in `core/src/trezor/ui/layouts/tr`
  - hopefully, both these modules could share general methods being called on `layout`, like `paint`, `timer` or `attach_timer_fn`

matejcik, 2022-11-14:
This should be done a little differently:
Rust should expose an API that is as unified as possible -- the ultimate purpose here is to get rid of the Python translation layer, and ideally make `trezor.ui.layouts` just a name for `modtrezorui2`.

(At this moment we still need to wrap the Rust layouts in a `RustLayout`. An intermediate solution might be to move the `interact()` calls to application code, so that instead of `await confirm_action(...)` we call `await interact(confirm_action(...))`. Another code- and RAM-saving option is to make a module level getattr like so:
```
def __getattr__(funcname: str) -> Callable[..., Awaitable]:
     rust_layout = getattr(modtrezorui2, funcname)
     def layout_call(*args, **kwargs) -> Awaitable:
           layout_obj = rust_layout(*args, **kwargs)
           return interact(RustLayout(layout_obj))
```
)

To be able to do everything we need in Rust, we also need to expose the ability to call ButtonRequests from inside Rust -- presumably via a callback similar to `timer_fn`.

After this, only the functions that actually have significant differences would go through a Python translation layer, and call into their own modules `modtrezorui_tt` and `modtrezorui_tr`. Most of the functionality should remain in `modtrezorui` though.

obrusvit, 2024-11-11:
Referring #4322