trezor/trezor-firmware #2636: TR design - PIN, wipecode

Hannsek, 2022-11-14:
[image]

grdddj, 2022-11-14:
Nice, I like that we are showing the prompt and trials left in the missing header - I currently have a screen before showing this info.

With the `CHECK PIN: please re-enter`, that might not be necessary right now, as the prompt will say `Re-enter PIN`.

Just to make sure, I am starting the numbers from 0.

Note to myself - I need to update the `eye` icon, I have the old version

Hannsek, 2022-11-14:
`CHECK PIN: please re-enter` is mandatory from the user POV.

Starting the numbers from 0 is fine.

grdddj, 2022-11-14:
Current state of the PIN setting after [73458cb](https://github.com/trezor/trezor-firmware/pull/2610/commits/73458cb407314f8994bca5b3722babf1e0019980):

[image]

Hannsek, 2022-11-15:
114, 115 - copy is bold, missing "PIN" in copy -> _Please re-enter PIN to confirm_

Hannsek, 2023-03-09:
Changed copy on PIN screen: 
[image]
It should be at least 4 digits long --> PIN should be 4-50 digits long.

grdddj, 2023-03-09:
Currently looks like this:

[image]

Is it alright, or do we want to make it pixel-perfect with Figma?

Hannsek, 2023-03-09:
The letters seems off. Also the positioning of the paragraphs is weird. Can we make it like this?
[image]

grdddj, 2023-03-09:
I could do this with minimal changes:

[image]

It currently uses the general way of rendering text, but we should be able to create a specific, "hard-coded" version just for this flow. Will try it

Hannsek, 2023-03-09:
We should get rid of the first two arrows.

closed 2023-05-31 by grdddj