trezor/trezor-firmware #2637: Add anti-klepto to signing

prusnak, 2022-11-14:
Although Trezor code is fully open-source and builds are fully deterministic, this article provides good reason why to include the anti-klepto protocol to sec256k1 signing: https://shiftcrypto.ch/blog/how-almost-all-hardware-wallets-can-steal-your-seed/

Currently, the blocker is that this protocol currently does not support Schnorr signatures (=no Taproot), but this is to be addressed here: https://github.com/bitcoin-core/secp256k1/pull/1140

Once the above issue is resolved and the code is merged into the secp256k1 we might consider adding the anti-klepto feature.

jadzeidan, 2025-03-17:
What about adding anti-klepto for non-taproot accounts in the meantime?

h0us3c4ll, 2026-10-27:
Good idea to have ... any update , @prusnak ? Thanks !

andrewkozlik, 2026-10-27:
It's in the works, see https://github.com/trezor/trezor-firmware/pull/4826. We hope to get it done in the upcoming months. Unfortunately there are higher priorities, so I don't want to make any promises about the delivery date.

h0us3c4ll, 2026-10-28:
> 8\. It's in the works, see [#4826](https://github.com/trezor/trezor-firmware/pull/4826). We hope to get it done in the upcoming months. Unfortunately there are higher priorities, so I don't want to make any promises about the delivery date.

thank you !