trezor/trezor-firmware #2638: QR code of Taproot account can't be scanned from the T1 device

STew790, 2022-11-14:
**Describe the bug**
QR code for Taproot address shown on T1 can't be scanned on Android devices (works for iPhone). It works for the production version of the T1 firmware but not with the latest master T1 version.

**Steps to reproduce:**
1. Go to Taproot account in Suite
2. Click on "Receive"
3. On device, press "QR code"
4. Try to scan QR code from T1 on Android device.

**Info:**
 - Suite version: desktop 22.12.0 (ff01d767b3827ce73296a25fa36cfded68457074)
 - Browser: Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/22.12.0 Chrome/100.0.4896.160 Electron/18.3.5 Safari/537.36
 - OS: Linux x86_64
 - Screen: 1920x1080
 - Device: model 1 1.11.3 Universal (revision 1852059e87d2070239cbfadb11b8ca516f80457c)
 - Transport: bridge 2.0.32

**Expected result:**
QR code of Taproot account can be scanned on T1 device.

**Actual result:**
QR code of Taproot account can't be scanned on T1 device.

matejcik, 2022-11-14:
how many different android devices and QR code scanner apps have you tried? :)

STew790, 2022-11-14:
Sorry, should have been more specific. We tried it on 2 Android devices (11 and 12). Tried it with Exodus app, our Trezor Suite in Chrome and a standard Android camera app. 

But @bosomt just tried to read it on MacOS via https://webqr.com/index.html and it didn't work for him there.
And I tried it with a standard iOS iPhone SE 2020 camera and it registered it immediately.

Did it work for you somewhere?

matejcik, 2022-11-14:
i believe @grdddj did test this on Android, but I don't know what app / phone / whatever.

our options now are:
(0) ignore this, say that Android QR libraries need to step up their game
(1) show taproot addresses in uppercase, which makes the QR code smaller (Suite will dislike us because they'll need to show tr addresses in uppercase for T1 but lowercase for TT)
(2) cut off the other part of the QR code. Maybe Android libraries dislike their pixels missing on one side but won't mind the other. We could also _rotate_ the QR code and cut off one of the sides.
(3) Make the blocks smaller -- use same density of data, but squish it into a half-size image. This will be scannable _if_ the phone's camera is good enough to focus on the smaller image.

We should test out (2) and (3), and keep (1) as our fallback option.

matejcik, 2022-11-14:
fwiw, the ZXing Android app also fails to scan this in the correct orientation, but it scans fine when I turn the phone sideways.
ZXing's library is relatively popular so this might explain parts of it.
The ZXing Decoder Online can't read it even if I rotate the image sideways though so who knows :woman_shrugging: 

(Samsung's Camera app has no problem and neither does QRBot, which is a 3rd party app I picked at random)

grdddj, 2023-02-22:
> our options now are:
> (0) ignore this, say that Android QR libraries need to step up their game
> (1) show taproot addresses in uppercase, which makes the QR code smaller (Suite will dislike us because [...]

It is strange that it was working in the past, both for me and the QA - https://github.com/trezor/trezor-firmware/issues/2190#issuecomment-1252495019 - QR readers on Android could have gotten more restrictive.

In the end, I did (3) in https://github.com/trezor/trezor-firmware/pull/2842 - even when being smaller, my `Android 13` can scan it nicely (now...).

Some users with older phones might still complain, the best-user-focused-solution would be probably (1), but probably not worth the complications on the Suite side.

Could not try it on a hardware device however, I was getting `Hard fault` repeatedly after installing the firmware.

mmilata, 2023-02-22:
> Could not try it on a hardware device however, I was getting `Hard fault` repeatedly after installing the firmware.

Are you building it with [`PRODUCTION=0`](https://docs.trezor.io/trezor-firmware/legacy/index.html#how-to-install-custom-built-firmware)?

grdddj, 2023-02-22:
> > Could not try it on a hardware device however, I was getting `Hard fault` repeatedly after installing the firmware.
> 
> Are you building it with [`PRODUCTION=0`](https://docs.trezor.io/trezor-firmware/legacy/index.html#how-to-install-custom-built-firmware)?

Thanks, that was the reason.

Now I can confirm I can scan it also on the hardware, even though it is not immediate, it takes some time for the QR code to be focused

Hannsek, 2023-02-22:
I can scan it right away..

closed 2023-02-22 by grdddj

bosomt, 2023-02-24:
Tested on Redmi Note 10 and Galaxy Tab S7 FE both times i was not able to scan new Taproot QR , both time it was too small to focus via internal camera

grdddj, 2023-02-24:
> Tested on Redmi Note 10 and Galaxy Tab S7 FE both times i was not able to scan new Taproot QR , both time it was too small to focus via internal camera
> 
> EDIT: i was able to scan QR code via both Android devices after i went out to get more light :)

Quite mixed results :) Let probably product decide if this is alright (@Hannsek) - if not, I am afraid we would need to contact `Suite` (or not care about the mismatch between Trezor (uppercase) and Suite (lowercase)

Hannsek, 2023-02-24:
It is alright from my POV.

bosomt, 2023-02-26:
retested successfully  on 3 devices and approved issue