trezor/trezor-firmware #2640: Quicken device tests by optionally disabling repotr generation after each test

grdddj, 2022-11-14:
As discussed in https://github.com/trezor/trezor-firmware/pull/2580#discussion_r1020338696 and the reply, the device tests are being slowed down by generating an HTML report after each individual test-case.

We can do something about it, the easiest is probably having a `pytest` option disabling this and using in in our `CI` runs.

matejcik, 2023-11-28:
is this maybe already resolved @grdddj ?

grdddj, 2023-11-28:
> is this maybe already resolved @grdddj ?

I think it is not, the HTML reports are still generated after each test (which has advantages locally, but not in CI).

The priority of this seems very low, it explains why it has been sitting here for over a year

matejcik, 2023-11-28:
Which "html reports" do you mean?
Per-test HTML reports are of course generated after every test -- even if we change that, it doesn't matter for CI because the total time will be the same regardless of _when_ you generate them.
`index.html` is indeed regenerated after every case, but that is very fast
`differing_screens.html` are only generated after the whole test suite finishes -- `testreport.generate_reports()` only runs in `pytest_sessionfinish`, the `teardown` function doesn't exist anymore

matejcik, 2023-11-28:
> `index.html` is indeed regenerated after every case, but that is very fast

not even that! it's regenerated ondemand by `show_results.py` independent of the test suite progress

closed 2023-11-28 by matejcik

grdddj, 2023-11-28:
> Which "html reports" do you mean? Per-test HTML reports are of course generated after every test -- even if we change that, it doesn't matter for CI because the total time will be the same [...]

Right, I agree. Thanks for analyzing and closing this