trezor/trezor-firmware #2651: Softlock behaves incorrectly with CoinJoin

matejcik, 2022-11-22:
from Slack:

> Apparently the soft lock during coinjoin behaves somewhat unexpectedly. I left the device alone with lock time 10 min (afaik default) and after few hours, the device is still not locked. Looks like the API calls around coinjoin are resetting internal counter, which they shouldn't IMO.

Presumably the device should not be usable, but CoinJoin messages should be allowed to run through.

With current softlock implementation any USB message will reset the timer.
We might want to differentiate "soft" lock touch (done by an USB message) from a "hard" touch (done by user interaction).
A "hard" touch resets the internal timer. A "soft" touch keeps the timer running, but if there are any "soft" touches registered at the end of the period, the locking will not shut down the current workflow.

To test: CoinJoin operations (via `DoPreauthorized`) are allowed to run even if the device is currently locked. Does this actually work, i.e., doesn't the signing process somehow implicitly rely on storage being unlocked?

hynek-jina, 2022-12-02:
This needs to be fixed before Trezor Suite 23.2.1 release

andrewkozlik, 2022-12-05:
> With current softlock implementation any USB message will reset the timer. We might want to differentiate "soft" lock touch (done by an USB message) from a "hard" touch (done by user interaction). A [...]

Makes sense.

> To test: CoinJoin operations (via `DoPreauthorized`) are allowed to run even if the device is currently locked. Does this actually work, i.e., doesn't the signing process somehow implicitly rely on storage being unlocked?

IIRC we have a device test for this and I tested it manually too. It uses `APP_COMMON_SEED` in the session cache.

mmilata, 2023-04-05:
Tested this with Suite, the idle timer is woken up by `SetBusy` and workflows starting with `DoPreauthorized`. Btw the coinjoin proceeds fine even when device is locked.

The idle timer is touched from following places (plus user interaction):
* [workflow.py](https://github.com/trezor/trezor-firmware/blob/c21cf87c53f79ad8de9d0549bb714cd08f9b46a0/core/src/trezor/workflow.py#L38), at the start of each workflow
* [wire/\_\_init\_\_.py](https://github.com/trezor/trezor-firmware/blob/c21cf87c53f79ad8de9d0549bb714cd08f9b46a0/core/src/trezor/wire/__init__.py#L197), in `read` and `read_any`, i.e. every time a message is decoded inside a workflow
* [apps/base.py](https://github.com/trezor/trezor-firmware/blob/c21cf87c53f79ad8de9d0549bb714cd08f9b46a0/core/src/apps/base.py#L372), in `reload_settings_from_storage()` called from `boot()`, i.e. every time a workflow restarts

The last one I don't know what to do with it, if I don't restart the timer I lose it entirely due to session restart, there don't seem to be a way to get hold of the old deadline?

matejcik, 2023-04-05:
> The last one I don't know what to do with it, if I don't restart the timer I lose it entirely due to session restart, there don't seem to be a way to get hold of the old deadline?

true, that is not tracked at all currently. ISTM you could just place the value as a global in `storage/cache.py`, that should persist through the restart? and then bump it every time in addition to extending the deadline

PR #2933 (2023-04-07): core: prevent USB events from waking autolock timer
~~Given the timeline and complexity I did not feel confident implementing the proposed solution of #2651. This PR introduces different solution which avoids touching the idle timer when processing message listed as allowed to be processed when device is locked.~~

~~The disadvantage is that with unfortunate timing the autolock interrupts coinjoin signing if there's one in progress.~~

Fixes #2651.
closed 2023-04-07 by mmilata