trezor/trezor-firmware #2652: TR design - Receive

Hannsek, 2022-11-22:
[image]

grdddj, 2023-01-02:
Done in [7926b02](https://github.com/trezor/trezor-firmware/pull/2610/commits/7926b02a44ef99ff97b4242cae9cf3c1c244ec05)

As we discussed, `xpubs` in case of multisig address are currently not supported

Hannsek, 2023-01-30:
[image]

@grdddj add header with currently used wallet

Hannsek, 2023-02-10:
[image]

Hannsek, 2023-03-29:
[image]
Please change _trezor.io/support_ to bold.

grdddj, 2023-03-29:
> Please change _trezor.io/support_ to bold. 

Done in [c2c9156](https://github.com/trezor/trezor-firmware/pull/2610/commits/c2c9156ad06435d2dacbe6ab018dc95758733f55)

Hannsek, 2023-05-15:
In case of multisig -> put this as an header: 
[image]

closed 2023-05-31 by grdddj