trezor/trezor-firmware #2656: TT UI2 scrollbar

Hannsek, 2022-11-23:
[image]

Max 7 visible dots. Max 5 big dots at the ends, otherwise 3 big dots.

big - 100 %, middle - 50 %, small - 25%

Figma: https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Trezor-T%2C-Trezor-One?node-id=1126%3A237&t=1RfM47z4HRCL9v6P-0

closed 2022-12-07 by mmilata

bosomt, 2022-12-13:
**QA OK**

scrollbar works/looks like expected.

Noted that text at last page is centered in middle of the screen, not sure if its expected or not.
I can create follow up issue if not expected.

[image]

**Info:**
 - Suite version: desktop 22.12.1 (ddaacdb2245d7a909b3d2abfe918cff34d2e1609)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/22.12.1 Chrome/104.0.5112.124 Electron/20.3.5 Safari/537.36
 - OS: MacIntel
 - Screen: 1440x900
 - Device: model T 2.5.4 Universal (revision ff35c77bad82c7acac1df6245940ebe0d6454642)
 - Transport: bridge 2.0.32

Hannsek, 2022-12-13:
Are the two dots on the edges smaller than the 3 dots in the middle?

Hannsek, 2022-12-13:
Have you tried more than 5 pages? Did you see smaller dots?

bosomt, 2022-12-14:
@Hannsek as soon i had more than 5 pages i see that size was changing as i scrolled

PR #2670 (2022-12-07): Very long scrollbars on Model T
Fixes #2656. [[UI diff](https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/3427058679/artifacts/master_diff/index.html)]

[image]