trezor/trezor-firmware #2658: Multisig option when exporting public key

sime, 2022-11-23:
Related issue: https://github.com/trezor/trezor-suite/issues/6918
Slack discussion: https://satoshilabs.slack.com/archives/C02V2PSDNA2/p1669125576505009

Introduce an option in CTL in `get-public-node` and in protobuf to export multisig specific xpubs.

CaveRock, 2022-11-30:
Is the slack discussion public? Would like to get an idea on possible timeline for fix to [issue 6918](https://github.com/trezor/trezor-suite/issues/6918) as this has effect on setting up a multisig wallet in electrum using public keys pulled at BIP48 paths.

sime, 2022-11-30:
@CaveRock It's not public and cannot offer a timeline.

Though https://github.com/trezor/trezor-suite/issues/6918 doesn't affect Electrum, Electrum uses`trezorlib`Python library, which also lives in this repository.

CaveRock, 2022-11-30:
@sime 

Thanks for the feedback.

Just to give some more info on our electrum issue, which may not be a major use case and I can understand there not being an eta on an update but for interest sake I wanted to describe it.

The Electrum problem we have is where a watch only Electrum native segwit multisig wallet is being setup using master keys pulled from Trezor devices using Trezor Connect. 

If a user uses Trezor connect to pull keys at BIP48 path to setup a watch only Electrum native segwit multisig wallet, the keys the Trezor will export will be vpubs (zpubs). If one attempts to submit a vpub (zpub) as the master key in Electrum native segwit multsig wallet setup it will, correctly, throw an error. 

This means that the user must know to convert the vpub (zpub) to a Vpub (Zpub) before inputting into Electrum to create the watch only wallet.

It also means that there is no way for a user to have their Trezor confirm a Vpub (Zpub) on screen for peace of mind in confirming they are part of a native segwit multisig wallet.

(I have put main net prefixes in brackets for clarity)

sime, 2022-11-30:
Thanks for the context. You are using Trezor Content medium for fetching master public keys which will be input into Electrum.

So no direct Electrum to Trezor integration.

CaveRock, 2022-11-30:
That's correct, and I also confirm no direct Electrum / Trezor integration.
Thanks.

matejcik, 2022-11-30:
> It also means that there is no way for a user to have their Trezor confirm a Vpub (Zpub) on screen for peace of mind in confirming they are part of a native segwit multisig wallet.

FWIW, this _can_ be done. The way to do it is to show a _receiving address_, then you can confirm all the multisig cosigners, including yourself, with the correct Vpub / Zpub prefixes.
The missing step is of course fetching the public key before you have all the cosigners together

CaveRock, 2022-11-30:
@matejcik 
Ya, we have used the confirm address feature that can be used with Unchained Capital library for giving customer peace of mind for checking their wallet addresses and the feature works well! And as you say, it just the missing step on wallet setup itself, but having the address checked does mitigate this somewhat. Thanks

gwilkinson01, 2024-11-01:
So possibly related to this issue: I created a multisig native segwit wallet in Sparrow. Added a keystore using my trezor safe 3 (firmware: 2.8.0). I wanted to verify that the zpub shown in sparrow matched with the one shown on the trezor device so I ran this command: trezorctl btc get-public-node -n "m/48h/0h/0h/2h" -d but the zpub that was shown on the trezor did not match with that in sparrow. 

Retrieving the xpub for the legacy path `m/45'` worked perfectly.

maybe also relates to https://github.com/trezor/trezor-suite/issues/6918

andrewkozlik, 2024-11-02:
@gwilkinson01 when you use the trezorctl command, does Trezor show a zpub or an xpub? How about if you add `-t segwit`:
```
trezorctl btc get-public-node -n "m/48h/0h/0h/2h" -d -t segwit
```

gwilkinson01, 2024-11-02:
> andrewkozlik

The Trezor shows a zpub (although the confusingly the header states xpub 😆). See below image:

[image]

I get the same output with `trezorctl btc get-public-node -n "m/48h/0h/0h/2h" -d -t segwit`

andrewkozlik, 2024-11-02:
@gwilkinson01, I see. Your problem is indeed what this issue is about. I think that a simple workaround for now is to use `-t address` with `trezorctl`. That should force Trezor to show an xpub on the screen. Sparrow lets you switch between the "Zpub" prefix and the "xpub" prefix, so that you can compare them:
[image]

Also @matejcik proposed another workaround [above](https://github.com/trezor/trezor-firmware/issues/2658#issuecomment-1331962625).

gwilkinson01, 2024-11-02:
@andrewkozlik perfect, works for me. Thanks for the swift response!

andrewkozlik, 2024-11-03:
FYI, I created a draft PR to fix this, see https://github.com/trezor/trezor-firmware/pull/4305. I still need to implement the corresponding changes in legacy and add device tests.