trezor/trezor-firmware #2664: TR design - Send

Hannsek, 2022-11-29:
[image]

Hannsek, 2022-12-19:
[image]

grdddj, 2023-01-01:
Without the final "Transaction Signed" `Popup` done in [44490e6](https://github.com/trezor/trezor-firmware/pull/2610/commits/44490e64bcb6d7592dd6853c35a943fcfff8b97d)

[image]

matejcik, 2023-01-12:
"Recipient #1" ends with a "Confirm", but then the "Amount #1" allows going back?
that seems weird. if we do allow going back (reasonable), then at least the verb should not be "confirm" because at that point it's not really confirmed

Hannsek, 2023-01-13:
There was a discussion about that and according to them this is the best wording…

Hannsek, 2023-01-30:
[image]

Here is slightly bigger `i` sign than in receive flow. (See figma https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Trezor-T%2C-Trezor-One?node-id=5437%3A36557&t=Iu3ljOwwRQscCic4-4)

Different button on `recipient` tab - `Continue` instead of `Confirm`

Hannsek, 2023-02-21:
[image]

closed 2023-05-31 by grdddj