trezor/trezor-firmware #2667: PIN delete button (yellow back arrow)

Hannsek, 2022-11-30:
**Describe the bug**
PIN entry page.
When user clicks on the yellow back arrow, it deletes all already typed characters, not only one.

**Firmware version and revision**
2.5.2

**To Reproduce**
Steps to reproduce the behavior:
1. type several numbers on PIN entry page
2. press yellow back arrow
3. It deletes all entered numbers

**Expected behavior**
It should delete only one number.

sime, 2022-11-30:
@Hannsek I think this is already resolved in UI2.

e.g. https://github.com/trezor/trezor-firmware/pull/2227

@mmilata Could you confirm?

matejcik, 2022-11-30:
yeah this is a duplicate of https://github.com/trezor/trezor-firmware/issues/1469

closed 2022-11-30 by Hannsek