trezor/trezor-firmware #2680: UI2: send transaction style update

mmilata, 2022-12-13:
[image]

Hannsek, 2022-12-13:
Figma file here: https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Trezor-T%2C-Trezor-One?node-id=4081%3A19006&t=VxjZTbyYG85Yz2AV-1

Hannsek, 2023-01-20:
New design is coming. For now, this is blocked.

mmilata, 2023-02-06:
Is this still blocked or is the figma design ready now?

Hannsek, 2023-02-06:
Still not ready, I hope it will be ready next week…

Hannsek, 2023-02-09:
[image]
[image]

Figma file: https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Trezor-T%2C-Trezor-One?node-id=6276%3A46984&t=fVkX6Rym50TQFKbi-4

Design is ready @mmilata. For now we won't use the chunks when displaying address. Later on we will use it.

andrewkozlik, 2023-02-15:
In the `recipient #1` screen:
- I think showing the currency in the account label is open for discussion. It should be the same as the currency that we are spending. If it's not, e.g. we are sending LTC to a BTC path, then it seems more appropriate to show the path, than trying to assign a name to it.
- I think we really need to use the word "account" in the account label. `L. SegWit #223` just seems unclear what it means, i.e. `L. SegWit account #223`.

Hannsek, 2023-02-16:
I agree with both of your points. 
[image]

sending LTC to a BTC path is allowed only behind safety checks, right?

andrewkozlik, 2023-02-16:
> sending LTC to a BTC path is allowed only behind safety checks, right?

Probably yes, but I think there might be some coin combinations which allow it even without safety checks. Those that have fork ID defined may allow it (Bitcoin Cash, Bitcoin Gold, Bitcoin Private). But my point is that it doesn't seem to make sense to assign a name to the account in that case, because the user will not see the incoming transaction in their Bitcoin account. If there is a coin name in the account label, then it should represent the network, not the path. (I think in the T1 implementation it actually represents the path, which is why I am saying it's open for discussion, because I don't really understand the rationale for that.)

Hannsek, 2023-02-27:
[image]

closed 2023-04-05 by Hannsek