trezor/trezor-firmware #2686: Delete unused components from micropython UI - Popup, Loader, Pageable...

grdddj, 2022-12-15:
`core/src/trezor/ui/loader.py` and `core/src/trezor/ui/popup.py` contain UI components that are not used anymore (because of moving UI to `Rust`). Also `core/src/trezor/ui/components/common/confirm.py` has unused `Pageable` class.

Still having them in codebase is just increasing the binary size.

grdddj, 2022-12-15:
Turns out the `Loader` is still being used via `LoadingAnimation` in https://github.com/trezor/trezor-firmware/blob/550e2a79512b75f09e57878baaa52bb5310aefbe/core/src/apps/management/reset_device/__init__.py#L26-L28

@mmilata Did we miss this when moving into `Rust` loaders, or is this intentional?

mmilata, 2022-12-15:
You're right, I have missed it. Will move it to rust side.

grdddj, 2022-12-15:
Thanks, I have tagged you (and this issue) in https://github.com/trezor/trezor-firmware/pull/2688, as after we remove `Loader`, there will be more removing to be done

closed 2023-01-25 by matejcik