trezor/trezor-firmware #2693: Incompatibility of python sdk with `eth-brownie`

madlabman, 2022-12-19:
**Describe the bug**

Installation of `eth-brownie` alongside `trezor` package breaks `eth-brownie` because of conflicting packages `rlp` a.k.a `pyrlp` used by brownie and `simple-rlp` used by trezor. Both packages provides the module with the same name `rlp` which causes ambiguity and broken dependencies setup.

**Firmware version and revision**

```
eth-brownie==1.19.2
trezor==0.13.4
```

**To Reproduce**

Steps to reproduce the behavior:
1. Create new project
2. Run `pip install eth-brownie==1.19.2`
3. Run `pip install trezor==0.13.4`
4. Run `brownie console`
5. See error:
```
<...>
    class HashableRLP(rlp.Serializable):
AttributeError: module 'rlp' has no attribute 'Serializable'
```

**Expected behavior**

Brownie runs its command.

prusnak, 2022-12-19:
This sounds like a problem that should be reported to brownie since trezor works as expected.

prusnak, 2022-12-19:
This sounds like a problem that should be reported to brownie since trezor works as expected.

@matejcik is there any way around this? (as in what should brownie fix on their side?)

madlabman, 2022-12-19:
Hey, @prusnak! Both projects work as expected without each other. I've reported here because it's much more possible imho for `trezor` project e.g. to move to `pyrlp` project. The alternative way is to ask `simple-rlp` maintainer to avoid using `rlp` as module name but it will require updates from `trezor` side as well.

prusnak, 2022-12-19:
> imho for `trezor` project e.g. to move to `pyrlp` project.

We will not move to pyrlp project, because we've already been there and we only need simple-rlp.

matejcik, 2022-12-19:
The reason for moving away from pyrlp is that pyrlp has a dependency on `eth-utils` and `eth-utils` are very dependency-heavy, very particular about their preferred versions of things (like Cython even, iirc?) and have caused major dependency related headaches in the past.

I would consider moving simple-rlp to an extra `trezor[ethereum]` but this has the drawback that then `trezorctl ethereum *` doesn't work out of the box.

User-side, you should be able to get around this by manually uninstalling `simple-rlp` and installing `pyrlp` instead, Trezor code should work fine with either.
I don't know how to express this as an install requirement though.

Brownie-side, no idea honestly. Switch to simple-rlp also :)

prusnak, 2022-12-19:
> I would consider moving simple-rlp to an extra `trezor[ethereum]` but this has the drawback that then `trezorctl ethereum *` doesn't work out of the box.

I would prefer not doing this. This is an issue between `pyrlp`, `simple-rlp` and `brownie`. So let them figure it out without crippling our out of the box experience.

madlabman, 2022-12-19:
Thank you for your response! I agree that effortless for users of `trezor` is in priority. I also strongly support an idea of using less coupled dependencies. The sad thing is partially compatible API of `rlp` replacement library in that case. But since @matejcik mentions that `trezor` is compatible with `pyrlp` we have a workaround at least.

closed 2022-12-19 by matejcik