trezor/trezor-firmware #2694: T2B1 - Shamir backup

Hannsek, 2022-12-19:
[image]

grdddj, 2023-01-05:
Shamir and SuperShamir wallet creation (not recovery yet) roughly done in [0a60d31](https://github.com/trezor/trezor-firmware/pull/2610/commits/0a60d31c813f82f3884a144d1f61b62f354d2867)

[image]

SuperShamir:

[image]

matejcik, 2023-01-12:
will T2B1 have the "Info" button(s) in reset / recovery?

Hannsek, 2023-01-13:
The info is already in the flow by default. e.g. 
[image]

matejcik, 2023-01-13:
that's for Shamir backup. the TT info is more verbose, i.e.:
[image]
(subtitle about selected number)
[image]
(example with currently selected number)

I suppose that keeping it shorter on T2B1 makes sense

SuperShamir recovery info seems missing completely though?
[image]
[image]

Hannsek, 2023-06-02:
We can put "full info" under the `i` icon:

[image]