trezor/trezor-firmware #2696: T2B1 design - passphrase entry

Hannsek, 2022-12-19:
[image]

[image]

Hannsek, 2023-01-06:
[image]

Hannsek, 2023-01-24:
[image]

Hannsek, 2023-02-03:
missing `*`, `#`

+ change middle button from `confirm` to `select` when on `SHOW`

+ when the first character is space, `x cancel` do not change to `<- delete`

Hannsek, 2023-03-29:
[image]
header not displaying last `E` properly.

grdddj, 2023-03-29:
> [image] header not displaying last `E` properly.

Good catch, there is an "invisible" scrollbar at the right side. Fixed in [0098290](https://github.com/trezor/trezor-firmware/pull/2610/commits/0098290dcda3d5f9f06a0316c3c1c8a4f708669f)

closed 2023-05-31 by grdddj