trezor/trezor-firmware #2700: Missing passphrase dialogue on device display in UI2

bosomt, 2022-12-20:
**Describe the bug**
I'm missing passphrase dialogue on device display in UI2. See comparison with 2.5.3 public version

**Firmware version and revision**
 - Device: model T 2.5.4 Universal (revision 305b7fe84d802db7f0826d07a222a886f73732f0)

**Info:**
 - Suite version: desktop 23.1.0 (38b40ef729793e945c3ad2c1f45e29db49460311)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/23.1.0 Chrome/104.0.5112.124 Electron/20.3.5 Safari/537.36
 - OS: MacIntel
 - Screen: 1440x900
 - Device: model T 2.5.4 Universal (revision 305b7fe84d802db7f0826d07a222a886f73732f0)
 - Transport: bridge 2.0.32

**Screenshots**

_2.5.4_
[image] 

_2.5.3_
[image]

grdddj, 2023-01-23:
Actually solved by https://github.com/trezor/trezor-firmware/pull/2688/commits/4709e4e3ee56282953504b07af71f83c972f7a5b in https://github.com/trezor/trezor-firmware/pull/2688

closed 2023-01-25 by matejcik

bosomt, 2023-01-27:
**QA OK**

**Info:**
 - Suite version: desktop 23.2.0 (a7d89a929c5f22d6b80707269bd1d45900862dc1)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/23.2.0 Chrome/104.0.5112.124 Electron/20.3.5 Safari/537.36
 - OS: MacIntel
 - Screen: 1440x900
 - Device: model T 2.5.4 Universal (revision 461f5667777514e6921ec1a53ca3d023a2a6a092)
 - Transport: bridge 2.0.32