trezor/trezor-firmware #2706: Expose seed word count

sime, 2023-01-02:
## User Story
As a BIP39 user
I want to know how many words my device has
So I know how many words I need to write down

## Background
During onboarding/backup, we should prompt the user in Trezor Suite that they will have 12/24 words to write down/check.

## Unknowns
* Shamir: Is exposing the threshold a privacy leak?

matejcik, 2023-01-02:
Backup strength is selected by Suite, so it might already have this information. OTOH, if the device was initialized elsewhere, the information is not available.

Two options to do this:
(a) add this to `Features`
(b) add insert a message into the backup flow that tells you the information only once, during the backup itself.

> Shamir: Is exposing the threshold a privacy leak?

could be, but why should Suite need to know in the first place?

sime, 2023-01-02:
> Two options to do this:
> (a) add this to Features
> (b) add insert a message into the backup flow that tells you the information only once, during the backup itself.

My first idea was option (a) too. Though, option (b) doesn't help in test recovery scenarios.

> > Shamir: Is exposing the threshold a privacy leak?
> 
> could be, but why should Suite need to know in the first place?

The idea is to give as much context to the user as possible, _only if_ it doesn't leak sensitive information. e.g. Enter _threshold_ number of shares to test...

matejcik, 2023-01-02:
When recovering on TT, you select the number of words on Trezor screen and the host knows nothing about it. Also Trezor doesn't know the thresholds until you enter the first share.

We currently don't even have a mechanism for pushing this info back to the host, `ButtonRequest`s would not be suitable.

sime, 2023-01-02:
> When recovering on TT, you select the number of words on Trezor screen

Ah, I think you mean during seed check, you select the number of words. This functionality is intended for backup and seed checks only.

If seed word count were stored during seed creation and available via features, backup and seed checks flows could inform the user the number of words they will need to enter.

sime, 2024-08-02:
@Hannsek This is great to handle it on device w/ privacy leak to host either 💪

closed 2024-08-02 by sime