trezor/trezor-firmware #2709: UI2: tos - copy change

Hannsek, 2023-01-04:
Change copy for these kind of screens: 
[image]

**Delete the initial question.**
**Change heading**
**Change buttons**
[image]

https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Trezor-T%2C-Trezor-One?node-id=6015%3A45701&t=gdClSu39z7PYeyy2-4

matejcik, 2023-01-12:
the current screen reads "Shamir Backup" if shamir is in use.
do we completely drop that info?
[image]
[image]

Hannsek, 2023-01-13:
Good point, we can put it into the brackets on the new line of header: 
[image]

matejcik, 2023-01-13:
I would prefer not having multiline title, if we can at all avoid it. the layouts implicitly assume that the header is one line.
but @mmilata has a better idea about how disruptive that would be?

mmilata, 2023-01-13:
Not massively disruptive I think. We'd have to implement multiline Label<>s and then be careful not to have multiline label that changes content at runtime to something with different number of lines (titles can't be changed after constructor but labels can).

matejcik, 2023-01-13:
I did something that touches the possibility of multiline labels in the FIDO PR. Don't recall if I assume that it must be one line, or whether it depends on the preconfigured label height.

TychoVrahe, 2023-01-15:
we also have the marquee now, might be worth considering

PR #2748 (2023-02-17): core/ui: reset_device copy change
Fixes #2709.
UI diff: https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/3602753054/artifacts/master_diff/index.html

[image] [image] 
[image] [image]
closed 2023-02-17 by mmilata