trezor/trezor-firmware #2713: UI2: make sure the last layout before signing is always HoldToConfirm for all coins

grdddj, 2023-01-04:
Some coins like `Binance`, `EOS`, `Ripple` or `Tezos` do not currently call layout functions with `hold=True`, so there is no HoldToConfirm functionality, which we want

Hannsek, 2023-01-05:
related (original) issue: https://github.com/trezor/trezor-firmware/issues/2712

closed 2023-02-03 by grdddj

bosomt, 2023-02-10:
**QA OK**

during sending of TXRP you see `Hold to confirm` dialogue.

**Info:**
 - Suite version: desktop 23.2.1 (bcfbc9ffb59b51f6c7ecb58d529769ca0db2f7b1)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/23.2.1 Chrome/104.0.5112.124 Electron/20.3.5 Safari/537.36
 - OS: MacIntel
 - Screen: 1440x900
 - Device: model T 2.5.4 Universal (revision 9add2b0027048c617d780b4943c00718d72e8334)
 - Transport: BridgeTransport 2.0.32