trezor/trezor-firmware #2719: Enforce at least 4 digits PIN

Hannsek, 2023-01-06:
When PIN protection is enabled, PIN will have to be at least 4 digits long.

We won't allow user to continue with PIN shorter than 4 digits.

grdddj, 2023-01-26:
Do we have a concrete design of what and when we want to show users in this case their PIN is short?

There are at least two approaches:
1. The PIN dialogue itself will not allow for sending it when having less than 4 digits - could be showing some message in the PIN row(s) when attempting to enter it - e.g. "WARNING\nLess than 4 digits"
2. Allowing the PIN dialogue to send anything and then validating the length - and in case of failure displaying some individual dialogue with error message and after that going  back to the PIN dialogue

My preference is probably `1`, but we would need designs where to display what info - for both `TT` and `TR`.

matejcik, 2023-01-26:
would be nice if there was a visual indicator instead of text.
on TT this seems vaguely doable, something like, if there is no PIN, there is no OK button, and if there is 3 digits or less, the OK button is disabled and has some sort of "waiting" icon??
haven't seen the TR designs in quite some time, but maybe the confirmation button could say "enter more digits" or whatever

Hannsek, 2023-01-26:
I am also for the first option. On TT, the "enter" button can be greyed out. 
On TR, we will need to think about this.

I will post it here once it will be ready.

andrewkozlik, 2023-01-26:
Just to be clear, this should apply only when setting the PIN, not when entering the PIN in general, because otherwise we would lock-out users who already have a short PIN.

I'd also consider not restricting the user, but rather showing a warning screen if they enter less than 4 digits with the screen offering  [Back] and [Continue] buttons.

Does this also apply to the wipe code? Some users might want a single digit wipe code as a quick wipe feature.

Hannsek, 2023-01-26:
For model R, it can be used everywhere.

This does not apply to wipe code.